schreibfaul1/ESP32-audioI2S #2: Audio Brpppppp

danbicks, 2019-01-28:
Hi,

First of all awesome job with this library. I did have a bit of a pain installing it within Arduino IDE and had to create a separate folder with all the library files in for it to work.

Now works audio is really good, I am using a UD1344a DAC. When listening to a stream from time to time I get a brrrrrrrrrrr sound then audio resumes, seems possibly like a buffer issue. I have increased DMA-buffer from 64 to 128 and this has made no difference.

Any ideas on how this could be fixed, would make an awesome internet radio otherwise.

Big thanks

Dans

schreibfaul1, 2019-01-29:
Since Espressif ESp32 Version V1.0.1. there is in structure i2s_config_t the new descriptor "tx_desc_auto_clear". If true it helps in avoiding noise in case of data unavailability. That can be in case of weak datastreams. So install V1.0.1-rc4 , third party index url https://dl.espressif.com/dl/package_esp32_dev_index.json   and "set tx_desc_auto_clear" true (constructor in audio.cpp)
[image]

amakukha, 2019-05-26:
In my case, this could be solved by increasing buffer size to 1024, while decreasing number of buffers to 4.

schreibfaul1, 2019-09-05:
buffersize is now1024, buffercount is 4

closed 2019-09-05 by schreibfaul1