schreibfaul1/ESP32-audioI2S #3: More Examples

danbicks, 2019-01-28:
Hi,

Me again. Would it be possible to include more examples.
Playing a Mp3 or wave file from SPiffs and program mem would be awesome.

Well done again.

Super nice library.

Dans

schreibfaul1, 2019-01-29:
At this time a mp3 decoder is implemented. So it can play only mp3 from datasteams and from files, no wav

HillBillyWidgets, 2019-03-24:
Hi,
I am also playing around with the awesome library you wrote, and was wondering if there is a way to randomly play songs that are on the sd card automatically. I can't seem to get the audio.loop() to play more than one song on a SD card without a reset. I looked through the code, and don't see any option to loop randomly. Any idea or thoughts on how to do this would be greatly appreciated.

schreibfaul1, 2019-03-26:
After playing a mp3 file the event audio_eof_mp3 raises. Then play the next song.

````c++
//example
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
    static uint8_t idx = 0;
    idx++;
    if(idx==1) audio.connecttoSD("title001.mp3");
    if(idx==2) audio.connecttoSD("title002.mp3");
    ....
}
````

closed 2020-02-13 by stale[bot]

ProtoFoo, 2023-01-29:
> After playing a mp3 file the event audio_eof_mp3 raises. Then play the next song.
>
> ```c++
> //example
> void audio_eof_mp3(const char *info){  //end of file
>     Serial.print("eof_mp3 [...]

I am pretty sure this solution (from 2019) does not work like that anymore today (2023).

The `audio_eof_mp3` callback [is called](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L2815) inside the function [processLocalFile()](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L2695). That function is called inside the main [Audio::loop()](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L2237) while the loop still holds `mutex_audio` which was taken with `xSemaphoreTake()`.

Any new call to `audio.connecttoFS()` -- during the callback -- will never complete because it (correctly) [waits for `mutex_audio`](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L683) which is still held by `Audio::loop()` when the callback is executed.

Why am I sure about this? I modified the `I2Saudio` example to play one MP3 from SD card at the end of `setup()` and a second file in the `audio_eof_mp3()` callback. After perfectly playing the first file, the ESP32 waits forever [at the mutex](https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L683) to play the second file.

I don't have a good solution for this at the moment. It would be good to have a way to execute the callback after the mutex was released.

schreibfaul1, 2023-01-30:
I installed the semaphore to prevent double access ('thread safety') to the library in multitask environments. Of course, the own task should not be locked out in the event of a recursion.
Thank you for your contribution and the good error analysis. :-)