schreibfaul1/ESP32-audioI2S #4: MP3 decode error -3 / syncword not found

olup, 2019-03-31:
Hi
Kudos for the great work here. I don't understand everything when it comes to mp3 decoding, but i am running into a dead end with a file at this address : 
http://media.radiofrance-podcast.net/podcast09/10059-31.03.2019-ITEMA_22023210-0.mp3

It seems the decoder has trouble reading it, and that end up messing the frame detection, hence the syncword not found error ? I'll be thankfull for some explainations.

The last log says it finds a syncword at some position, then at zero, then panic.
Cheers !

schreibfaul1, 2019-04-03:
Hello olup,
http://media.radiofrance-podcast.net/podcast09/10059-31.03.2019-ITEMA_22023210-0.mp3 is not a usual mp3 radio stream, it is a podcast. Podcasts may have an ID3-Tag at the beginning. That's why the decoder can not found the next syncword (in this case the ID3 info is ~150k long). As a first workaround, the ID3-Tag is skipped.

olup, 2019-04-03:
This is extreemely interesting, I had no idea of the inner works of podcasts, need to go back digging. So, if I understand you correctly, you think the problem could be solved by trying to skip the ID3 tag of the begining, wich involve I guess first geting the tag size and then offsetting the read from there ? I have no idea on how to do that but i'll start looking. Cheers.

olup, 2019-04-03:
Ha ! you actually just commited those changes ! thanks a lot, i'll give it a spin in the morning !

olup, 2019-04-04:
A thousands thanks for this very quick update of the library, this is now working flawlessly. Thank you !

closed 2019-04-04 by olup