schreibfaul1/ESP32-audioI2S #22: ESP32 reboot

noevile, 2019-10-21:
I am creating a web interface for this wonderful library. But I ran into a reboot problem. 

String station;
audio.connecttohost (station);

As soon as the variable takes on a text value with the station address, the Internet radio starts to sound. Is this normal or a bug in the library?
Is it possible to find out if the audio stream is currently in progress?

Sorry for my English. I use google translator.

schreibfaul1, 2019-10-22:
If the webstream was interrupted (for some reasons) this library will try to reconnect to the last host after few seconds.

You can try this:

String station1="http://firstStation...");
String station2="http://nextStation..");
audio.connecttohost(station1);

...
some time later
...

audio.stopSong();

...
even later
...

audio.connecttohost(station2);

schreibfaul1, 2019-10-22:
Such function is not implemented, but you can query the flag m_f_running and see if it set or not

closed 2019-10-22 by noevile