schreibfaul1/ESP32-audioI2S #24: Playing files from SD

biologist79, 2019-11-04:
Inspired by the project "Tonuino" I'm actually about to port (so to say) this stuff to ESP32 in order to have wifi-support. Or at least I'm trying to do so :-) So I played around a bit with your lib which is working absolutely fine for mp3-webstreams. However, playing files locally from uSD basically works but leads to pretty distorted sound (tried two mp3-files @128 kBps and your mp3-file provided as well).

Set-up: ESP32 devboard (wroom-32)
I2S-DAC: Adafruit MAX98357A

Any suggestions?

biologist79, 2020-01-09:
Can be closed because the reason was the microsontroller. Don't know what's wrong with it, works with a new one.

closed 2020-01-09 by biologist79

ashtorak, 2020-06-14:
@biologist79  Are you the guy from https://www.mikrocontroller.net/topic/484042#6302484 ?
There issues with the wiring are mentioned. So it was the controller instead?

@schreibfaul1 Falls du Zeit hast, könntest du bitte mal in den Thread reingucken, den ich verlinkt habe? Ich habe da ein ähnliches Problem geschildert.

biologist79, 2020-06-14:
@ashtorak Das siehst du richtig, der bin ich :-) Ich kommentiere dann einfach auch im genannten Link.

ashtorak, 2020-06-14:
Hi. Danke dir schon Mal! Werd mir es morgen noch Mal alles genauer angucken.
Was war denn jetzt eigentlich bei dir mit dem Mikrocontroller? Du hattest ja auf GitHub geschrieben mit einem neuen wäre es dann gegangen. Oder war das doch nicht der Grund? Im Forumsbeitrag hast du nicht mehr Bezug drauf genommen.

biologist79, 2020-06-14:
Retrospektiv betrachtet muss ich sagen, dass das mit dem Mikrocontroller nix zu tun hatte. Es war tatsächlich ein Problem aufgrund der Jumperwires. Nach dem Auflöten hat es dann gepasst. Auch wenn man kurze Jumperwires verwendet, sind offenbar die Stecker dennoch ein Schwachpunkt.

ashtorak, 2020-06-14:
Alles klar! Diese Stecker sind echt nicht besonders robust. Da hatte ich auch schon den ein oder anderen schlechten Kontakt :D
Dann noch einen schönen Abend!

schreibfaul1, 2020-06-14:
Ich denke auch dass die Kabel das Problem sind. Mit langen Jumperwires habe ich schlechte Erfahrungen gemacht, besonders die SPI Leitungen müssen möglichst kurz sein. Und die gegenseitige Beeinflussung durch parasitäre Kapazitäten muss so klein wir möglich gehalten werden.
Viele Grüße aus Berlin und einen schönen Abend!