schreibfaul1/ESP32-audioI2S #25: ESP-32 PCM5012 with SD CARD play mp3

batuhansaribaba, 2019-11-15:
if(content.substring(1) == "8A 35 8C 1E") {audio.connecttoSD("/2.mp3"); }
   
    else if(content.substring(1) == "A6 78 C4 F6"){ audio.connecttoSD("/4.mp3"); }

Hello, I need it urgently. I am trying to play mp3 with rfid from SD card but the sound is slowly, please give a sample solution

schreibfaul1, 2019-11-15:
Hello,
I can not reproduce this. My environment is: 
ESP32DevBoard with 4MB RAM
no PSRAM
PartitionSheme: default 4MB with spiffs
ESP32 version 1.0.4-rc1
CPU frequency 240MHz

perhaps this will work:
````
SD.begin (SD_CS, SPI, 10000000);  //or 20000000 MHz
````
and make SPI wires short as possible

batuhansaribaba, 2019-11-15:
Didn't work, is there any other way? Thank you very much for your reply

schreibfaul1, 2019-11-15:
works the 320k_test.mp3 file in my folder additional_info?
Can I download a mp3 that is slow? Perhaps I have somethong to do if my library works incorrect (wrong bitrate, nr of channels etc)

batuhansaribaba, 2019-11-15:
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
#define I2S_DOUT      22
#define I2S_BCLK      26
#define I2S_LRC       25

Audio audio;

//String ssid =     "Wolles-FRITZBOX";
//String password = "*******";

void setup() {
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    SD.begin(SD_CS);
//    WiFi.disconnect();
//    WiFi.mode(WIFI_STA);
//    WiFi.begin(ssid.c_str(), password.c_str());
//    while (WiFi.status() != WL_CONNECTED) delay(1500);
    Serial.println("setuppp..");
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21
 audio.connecttoSD("/4.mp3");
 Serial.println("mp3 play..");
    //audio.connecttoSD("/320k_test.mp3");
    //audio.connecttohost("www.wdr.de/wdrlive/media/einslive.m3u");
//    audio.connecttohost("dg-ais-eco-http-fra-eco-cdn.cast.addradio.de/hellwegradio/west/mp3/high");
//    //audio.connecttohost("fischkopp.stream.laut.fm/fischkopp");
    //audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");
}

void loop()
{
    audio.loop();
}

// optional
//void audio_info(const char *info){
//    Serial.print("info        "); Serial.println(info);
//}
//void audio_id3data(const char *info){  //id3 metadata
//    Serial.print("id3data     ");Serial.println(info);
//}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
//void audio_showstation(const char *info){
//    Serial.print("station     ");Serial.println(info);
//}
//void audio_showstreaminfo(const char *info){
//    Serial.print("streaminfo  ");Serial.println(info);
//}
//void audio_showstreamtitle(const char *info){
//    Serial.print("streamtitle ");Serial.println(info);
//}
//void audio_bitrate(const char *info){
//    Serial.print("bitrate     ");Serial.println(info);
//}
//void audio_commercial(const char *info){  //duration in sec
//    Serial.print("commercial  ");Serial.println(info);
//}
//void audio_icyurl(const char *info){  //homepage
//    Serial.print("icyurl      ");Serial.println(info);
//}
//void audio_lasthost(const char *info){  //stream URL played
//    Serial.print("lasthost    ");Serial.println(info);
//}
//void audio_eof_speech(const char *info){
//    Serial.print("eof_speech  ");Serial.println(info);
//}

_______________________________   **very successful, running code.**

batuhansaribaba, 2019-11-15:
//-------------------------**code** with sound slow (sound **vibrating)**--------------------

#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

#include <SPI.h>
#include <MFRC522.h>

#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
#define I2S_DOUT      22
#define I2S_BCLK      26
#define I2S_LRC       25

Audio audio;

#define SS_PIN 21
#define RST_PIN 0

MFRC522 mfrc522(SS_PIN, UINT8_MAX );
 
void setup() {
  Serial.begin(250000);

  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
   SD.begin(SD_CS);   
   audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
   audio.setVolume(21);
   
  mfrc522.PCD_Init();

   

}

void loop() {

    audio.loop();
    

  mfrc522.PCD_Init(); 
//delay(250);

  if ( ! mfrc522.PICC_IsNewCardPresent())
  {
    return;
  }
  
  if ( ! mfrc522.PICC_ReadCardSerial())
  {
    return;
  }
  
  Serial.println("UiD tag :");
  String content = "";
  byte letter;
  for (byte i = 0; i < mfrc522.uid.size; i++)
  {
    Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
    Serial.print(mfrc522.uid.uidByte[i], HEX);
    content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
    content.concat(String(mfrc522.uid.uidByte[i], HEX));
  }
  Serial.println("");

  content.toUpperCase();
  
   
    if(content.substring(1) == "8A 35 8C 1E") {audio.connecttoSD("2.mp3");}
   
    else if(content.substring(1) == "A6 78 C4 F6"){ audio.connecttoSD("4.mp3");}
      

  

}

schreibfaul1, 2019-11-15:
mfrc522 and SDcard works on the same SPI, that can make some trouble, see
https://github.com/espressif/arduino-esp32/issues/1219
same, if you use a Catalex Adapter, see
https://raw.githubusercontent.com/schreibfaul1/ESP32-MiniWebRadio/master/additional_info/CATALEX%20MicroSD%20Card%20Adapter.GIF

closed 2020-02-13 by stale[bot]