schreibfaul1/ESP32-audioI2S #26: BCLK has to be touched for audio output

sirpete83, 2019-12-12:
Hallo,
I think it is not really a problem with the library. Maybe something with the wiring. But I get only sound while touching the BCLK PIN with something conductive. I thing there is a need of a capacitive load? 
I use MAX98357A Breakout board (Marking AKK NRB). 
ESP32 Board ist a Wemos LOLIN32.

Maybe someone can provide a solution for this. Thank you.

[image]

schreibfaul1, 2019-12-12:
Hi,
I also have the MAX98357A board and it works without problems. For BCLK, another port can be used, for example GPIO14. The connections LRC / BLCK / DOUT can influence each other, single short wires are better

sirpete83, 2019-12-12:
Hi,
I fixed the problem. As you mentioned the wires make the problem. There are reflections present on the wire caused be the wave impedance. An termination by an 100 Ohm resistor in series near to the source, in this case at the mcu board fixes the problem. At the moment I only terminated the BCLK wire, but it should be done for all three high speed data lines.

closed 2019-12-12 by sirpete83

chriscckan, 2023-08-01:
Hi, I connect my ESP32S3 board to the MAX98357A codec board, it shows the same problem.  Initially I use jump wires, it sometimes give an audio sound, sometimes does not, I think it could be wire bad contact so I change all wire to soldering, absolutely no more bad contact, but it still give no audio output, I use DSO to see the waveform on the pins, and I found that whenever my DSO probe touch the BCLK pin, sound output works, I known the board and software is both working but something is blocking the BCLK signal which is the highest freq amount 3 pins.  According to your solution of adding 100R to the BLCK pin on the source board, it works now.  Really thank you for your experience sharing.