schreibfaul1/ESP32-audioI2S #29: Adjust playback speed?

joymonkey, 2020-02-04:
Is it possible to adjust the playback speed of a specific MP3, so it plays a little faster or slower than the file was intended for? I'd like to be able to crudely change pitch.

schreibfaul1, 2020-02-04:
It is possible. Manipulate freq in sub Audio::setSampleRate()
give freq some offset

joymonkey, 2020-02-08:
Thanks, I'll see if I can give it a try.

On another note, I'm using a PCM5102 module (this one : https://amzn.to/38f4niE ).
I had some difficulty getting it to output without a lot of noise. For anyone else with this module, here's what I did...
PCM5102 DIN -> ESP32 IO25
PCM5102 LCK -> ESP32 IO26
PCM5102 BCK -> ESP32 IO27
PCM5102 XMT (Mute Control) -> 3.3V
PCM5102 FMT (Format Select) - > GND (this is the required connection to stop noisy output)

This will play MP3 from SD very nicely, though there is a small click-pop noise when the MP3 starts and ends. I'm guessing this is an issue with the module's hardware itself, and coupling capacitors on the output may help. To use this module without click-pop and without modification, we might also use an IO pin to drive the XMT pin high only when a file is playing.

mhier, 2020-03-19:
Interesting, I had to put FMT to HIGH. Maybe there are different modules out there.

closed 2020-05-25 by stale[bot]

pitlochry, 2023-08-05:
> It is possible. Manipulate freq in sub Audio::setSampleRate()
> give freq some offset

@schreibfaul1 and other who succeeded in changing the sampling rate: Sorry for this beginner-question, but how to do it?
In void loop(){ I set (from your example code 'plays all files in a directory')

```
    audio.setSampleRate(44100);
    audio.loop();
```
If I double the SampleRate I get double speed, but the sound is extremely choppy. What is the correct way to speed up audio? Thanks for any help.

schreibfaul1, 2023-08-05:
We have a µC in use, and that sets us limits. More than 1.5* speed is often not possible because the decoders cannot keep up. With `audio.audioFileSeek(speed)` the speed can be changed between 0.25 ... 1.5.

pitlochry, 2023-08-05:
Thanks for this information! When I use e.g. audio.audioFileSeek(1.4) in the loop before audio.loop(), the speed is faster than normal, but it is extremely choppy. So this is the expected behaviour? Or is the place where I set the speed wrong or is anything else wrong?