schreibfaul1/ESP32-audioI2S #39: Using external RAM as a buffer?

michelep, 2020-05-03:
Thanks for your work! It's really awesome. However, i often got a buffer underrun and i'm thinking if should be possible to use an external RAM via SPI like 23LC1024 for buffering

schreibfaul1, 2020-05-03:
Hello michelep,
For me this problem does not occur, I suspect that the buffer is not filled quickly enough, maybe the CPU frequency is not 240MHz?

michelep, 2020-05-08:
Thanks for your reply. I've solved moving audio tasks to CPU0, while mantaining all other tasks to the other core CPU1 :-)

michelep, 2020-05-08:
If you want, take a look to ESP32_WebRadio project on my repo ;-)

schreibfaul1, 2020-05-08:
Yes, gladly, a look at the web radio would be good, but an RPI display 320x480px is required.

michelep, 2020-05-09:
I prefer an old looking Nokia 5110 display with 4 capacitive buttons :)

reaper7, 2020-05-14:
> 
> 
> Thanks for your reply. I've solved moving audio tasks to CPU0, while mantaining all other tasks to the other core CPU1 :-)

@michelep - tnx for the tip! 
my m5stickc + PCM1502A very often interrupted playback from network stream (stations from this lib example), when I moved audio.loop to cpu0 the stream plays without clipping :)

schreibfaul1, 2020-05-14:
Perfect, the distribution of the tasks on both cores can be a good solution for some projects.

closed 2020-05-14 by schreibfaul1

schreibfaul1, 2020-05-18:
Didn't understand a word, but great video :-)
https://www.youtube.com/watch?v=sVKdvpsTuFA

michelep, 2020-05-18:
Thanks, my english is not so good and i'm a bit ashamed to make videos in that language