schreibfaul1/ESP32-audioI2S #41: Extending connecttoSD to connecttoFS

vvs551, 2020-05-07:
Could you extend connecttoSD to the more universal method for using SPIFFS and SD in the same time?

bool Audio::connecttoFS(fs::FS &fs, String sdfile){
...
    //fs::FS &fs=SD;

schreibfaul1, 2020-05-08:
I added the function connecttoFS(...)

jwktje, 2020-06-14:
Does this mean we could use audio files in the Flash of the ESP?

schreibfaul1, 2020-06-14:
Yes, it works, load an mp3 file in the SPIFF (eg with the Arduino IDE) and then play it: audio.connecttoFS (SPIFFS, "test.mp3");

jvet88, 2020-08-04:
This throws me an error: 'SPIFFS' was not declared in this scope

schreibfaul1, 2020-08-04:
At the beginning ````#include "SPIFFS.h````  
in setup() ````SPIFFS.begin();````
and later call e.g.   ````audio.connecttoFS(SPIFFS, "wobble.mp3");````

closed 2020-10-10 by stale[bot]