schreibfaul1/ESP32-audioI2S #44: Playing short mp3 audio clips

dmarc1234, 2020-06-12:
I have created an effects box to playback short sound effect files for use in radio productions. This box plays a sound clip when an associated button is pressed on the box. I am using a MAX98357A codec to drive the speaker and a ESP32-WROOM-32 board.

The issue I am experiencing is that sometimes the audio file is not played when the associated button is pressed and when I press the same button again it does playback. Additionally sometimes when I press a button and get no playback I then get this clip added to the next one played, especially if it a longer one.

I have the 'audio_info' enabled so I can see what is happening and the below output is shown when the button is pressed but no audio is generated on the speaker:

info        Reading file: click.mp3
info        ID3 version=3
info        ID3 framesSize=89
info        ID3 normal frames
info        stream ready
info        syncword found at pos 0
info        Channels=2
info        SampleRate=44100
info        BitsPerSample=16
info        Bitrate=320000
info        End of mp3file click.mp3

If I press the button again audio is then generated on the speaker with the same information displayed above.

Could this be a buffer problem due to the clips being so short and possibility the buffer not being flushed totally so part of a preceding audio clip could be left in the buffer when the next one is played. 
I have attached a couple of the clips to test, the click.mp3 being the one that produces the worst problem as its really short.

David
[clips.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/4770463/clips.zip)

schreibfaul1, 2020-06-13:
Hello David,
Thanks for the detailed error description.
I made this test scenario:
````
void audio_eof_mp3(const char *info){  //end of file
    static int i = 0;
    if(i==0) {i++; audio.connecttoSD("beep.mp3"); return; }
    if(i==1) {i++; audio.connecttoSD("beep.mp3"); return; }
    if(i==2) {i++; audio.connecttoSD("click.mp3");return; }
    if(i==3) {i++; audio.connecttoSD("click.mp3");return; }
    if(i==4) {i++; audio.connecttoSD("wobble.mp3"); return;}
    if(i==5) {i=0; audio.connecttoSD("wobble.mp3"); return;}
}
````

It turned out that click.mp3 was only played once. I have noticed that the last frame of the file is playing incompletely. And I immediately eliminated the cracking at the beginning.

closed 2020-06-13 by schreibfaul1

dmarc1234, 2020-06-15:
Hello

So to confirm is the issue the audio file or something happening in the buffer ?

David

dmarc1234, 2020-07-20:
Hello

Do you have any more feedback on this issue as it is becoming a real problem now that we are using the box more often.

Why during your test was the last frame playing incompletely, you never explained if this was and issue with the file or the playback.

schreibfaul1, 2020-07-20:
After the test scenario (from June 14th) the files wobble, click and beep are played in full. After the file has been read in full, the last frame is decoded and then all values ​​still available in the I2S memory are transferred. Only then does the EOF come. Unfortunately there is no way to read the status of the I2S memory on the ESP32 to determine whether it is empty. To do this, zeros are written in memory size to ensure this. This can cause a time delay at the end of the file.

dmarc1234, 2020-07-21:
Thank you for your reply.

I have downloaded the latest version and the audio playback is very much improved. However, the playI2Sremains() does now cause a problem with the repetition rate at which the clips can be played as this sub-routine blocks for a short period of time after the clip is played, as you have pointed out above. This is causing problems with other tasks in the program as I now have extended the hardware to provide a rolling LED display to show messages when we use the box in theatre productions and the display freezes for a short period of time after a clip is played.
Could the above sub-routine be made to clear the DMA buffer as part of the loop process, in a similar way as the playback buffer is filled, so that this time delay is prevented ?

schreibfaul1, 2020-07-21:
Hope this works better now :-)