schreibfaul1/ESP32-audioI2S #48: Problems with https port 443 connections.

maciq4you, 2020-07-01:
Great library! I am having issues when connecting to https (port 443). The stream gets interrupted every 30 sec or so. Any insights or possible solutions are appreciated. I tested this with demo code (I2Saudio.ino). Example link: `https://arabellaaustropop.stream.arabella.at/arabellaaustropop`

schreibfaul1, 2020-07-01:
For me it also takes a moment until the SSL connection is established. Arabella Austropop takes 12s. This is more of an issue for the Espressif Forum. It is better not to use https, especially since the SecureClient uses ~ 12KBytes of RAM. An alternative is `arabella.stream.kapper.net:8005/arabellaaustropop` However, 30 seconds is too long.

maciq4you, 2020-07-01:
Thanks! I thought that SSL might be the problem. For now I will try to find non https sources for my streams.

closed 2020-09-06 by stale[bot]