schreibfaul1/ESP32-audioI2S #50: ESP32-WROVER - additional buffer

buhahapl, 2020-07-18:
Hello.
Do you plan to enhance library to use additional 4 or 8 MB of PSRAM as an stream buffer on new ESP32 modules?
When the internet is stable everything is fine, but when the internet is unstable (e.g. while listening to the radio in a car or there are problems with network) there are a lot of buffer underruns. The same station on mobile phone or PC works perfectly.

thanks,
wk

schreibfaul1, 2020-07-18:
Hello,
The memories are deliberately kept small and have the size to process an (aac or mp3) frame. And it still has to be possible to control a display or  build a internal web server. I could increase the input memory and run it as a ring. Maybe that could be a solution.

buhahapl, 2020-08-07:
Hello, I bought ESP32 VROWER-B version with 8MB PSRAM and I have just implement circular buffer for data incoming from socket.
I derived new class from WiFiClient override read(uint8_t *buf, size_t size), read() and available() methods. New class was used for  'client' member in Audio class.
Now sound is not interrupted by not stable internet connection.
This module really kicks ass :)

wga22, 2020-08-14:
> Hello, I bought ESP32 VROWER-B version with 8MB PSRAM and I have just implement circular buffer for data incoming from socket.
> I derived new class from WiFiClient override read(uint8_t *buf, size_t size), read() and available() methods. New class was used for 'client' member in Audio class.
> Now sound is not interrupted by not stable internet connection.
> 
> But I found another issue with buffering https connection.
> When I run the code below and read bigger amount of data than requested by Audio class in "size" parameter, stream do not start playing and there is "Stream lost -> try new connection" message.
> When I read exactly "size" amount of data, everything is ok:
> `WiFiClientSecure::read(m_clientBuffer, **size**);`
> 
> I know that there are problems with WiFiClientSecure, but maybe you have any idea what wrong is happening here...
> 
> ## the code:
> ```
> // buffered secure client, derived from WiFiClientSecure class - works perfectly when derived from WiFiClient class!
> // buffer - circular buffer of 512kB PSRAM capacity
> // m_clientBuffer - 6144 bytes helper buffer in conventional RAM
> // m_clientBuffSize = 6144 
> 
> int BufferedWiFiClientSecure::read(uint8_t *buf, size_t size)
> {
>   int received = WiFiClientSecure::read(m_clientBuffer, m_clientBuffSize);
>   if(received > 0)
>     buffer.put(m_clientBuffer, received);
> 
>   return buffer.get(buf, size);
> }
> 
> int BufferedWiFiClientSecure::read()
> {
>   int received = read(m_clientBuffer, 1);
>   if(received == 1)
>     return *m_clientBuffer;
>   else
>     return -1;
> }
> 
> int BufferedWiFiClientSecure::available()
> {
>   return buffer.available() + WiFiClientSecure::available(); // buffered and 'to read'
> }
> ```
> 
> ## logs:
> info buffers freed, free Heap: 157992 bytes
> info Connect to new host: https://online.nasze.fm:8443/stream
> info Connect to online.nasze.fm on port 8443, extension /stream
> streaminfo Connect to online.nasze.fm on port 8443, extension /stream
> info SSL/TLS Connected to server
> info SSL has been established, free Heap: 114460 bytes
> info Server: Icecast 2.4.3
> info audio/mpeg seen.
> info format is mp3
> info MP3Decoder has been initialized, free Heap: 134052 bytes
> info Cache-Control: no-cache
> info Expires: Mon, 26 Jul 1997 05:00:00 GMT
> info Pragma: no-cache
> info Access-Control-Allow-Origin: *
> info icy-aim:N/A
> bitrate 128000
> info icy-genre:Misc
> info icy-icq:0
> info icy-irc:N/A
> station Nasze Radio
> info icy-pub:0
> icyurl http://www.nasze.fm
> info Switch to DATA, bitrate is 128000, metaint is 16000
> lasthost https://online.nasze.fm:8443/stream
> **info Stream lost -> try new connection**

Could you share this code?

buhahapl, 2020-08-14:
Sure, https://gist.github.com/buhahapl/af8f17f0da00095b2b889264928548f4

In your Audio class you should replace/add:

Audio.h
#include "BufferedWiFiClient.h"
BufferedWiFiClient      client;
BufferedWiFiClientSecure clientsecure;

Audio.cpp
bool Audio::connecttohost(String host){
    client.init(); // add this
    clientsecure.init(); // add this

Unfortunately i haven't BufferedWiFiClientSecure code at this moment but for https connections, you should create BufferedWiFiClientSecure class in the same way as BufferedWiFiClient but derive it from WiFiClientSecure.

that's all
thanks, wk

buhahapl, 2020-08-15:
I think circular buffer should be used in Audio class, not as extended WiFi clients.

schreibfaul1, 2020-08-15:
Yes, that can work well, I am currently testing a WROVER module with PSRAM. I would then like to take over the CircularBuffer class

schreibfaul1, 2020-08-18:
PSRAM is now supported. If detected, InputBuffer will be allocated in PSRAM. Thanks buhahapl for the source. I have adapted the circularBuffer to my project. It works preferred with pointer to save memcopy time. Most stations send 100..300KBytes in advance. So 300k buffersize is sufficient.

closed 2020-08-18 by schreibfaul1