schreibfaul1/ESP32-audioI2S #51: Synchronize multiple devices

Insomniac80, 2020-07-18:
Hi,

I am running multiple ESPs with 5102s as a multiroom audio. I start them all by 
audio.connecttohost(station);
send to the same mqtt topic, which makes them start almost simultaneous. 
Audio sending device is a Raspberry 3 using Darkice (1 sec buffer, prio 4) which runs fine for the first 20 minutes up to an hour. 
But sooner or later it starts to become asynchron, no matter if I stream Spotify, from the phone, or play a local file. 
The ESPs do not host anything besides WLAN, MQTT, OTA and the music stuff. Also, the loop() - task has the highest priority among my tasks.

For now I changed getAudioCurrentTime() to return the time as float, and as soon as the difference is bigger then 0.07 I call stopSong(), followed by a small delay and connecttohost() on all devices, so they play sinchronous again after ~ 1+ second.

This solution has two problems:
- all sound is off and lost for +1 second
- I have to check all devices in a short interval, to have a resync short after the delay occurs.
maybe I could check internal by comparing playtime with uptime, to see if I can find a gap and react faster..

However, I do not even half understand your code good enough to maybe find a better solution.
 
For me one way could be to have an option to just play silence if data is not received, f.e. fill music array blank. This way the devices would always play synchronous, no upcoming and hearable out-of-syncing, only very short missing of sound. Especially for podcasts a very nice solution.
Another way (maybe the best) could be, as I read the request about more buffer: more buffer also on 4mb devices. If they could buffer a few hundred milliseconds, the problem would occur "far" later. Mayber with an option to check the remaining buffer size, so shortly before an out of sync I could resync if necessary.
Or, if both is not an option, a callback which informs about every gap the moment it occurs, maybe followed by the information of it´s length, so a resync could occur very short after, without checking the playtime of all devices.

schreibfaul1, 2020-07-19:
If I understand correctly, there is a network with several clients (ESP32). Each ESP32 accesses the same audio file. When a read cycle is running, the file is locked for the other clients. My solution is: The server waits for the EOF of all clients at the end of the current audio file and then asks all clients to play the next audio file.

Insomniac80, 2020-07-19:
I forgot to mention Darkice is using IceCast 2, which is a Audio Stream Server (which is atm running as soon as the Raspi is booted). I do not know enough about networking to determinate how it is handled (broadcasted or client after client, no idea). All I know is it is used also for webradios which have lots of client connected.
This way I can f.e. stream a file from my Podcast Catcher to my Raspi by bluetooth, and the formentioned combination records the stream from the analog output and re-sends it as a mp3-stream (.mp3) over wifi.
I also send what is locally played, f.e. by Raspotify, which plays live from Spotify, local files using pygame, or every other device connected by bluetooth.

Your solution presupposes to get the length of every song before playing, then starting the service for as long as the song is, then connect all clients, then start to play. At the end of the time (and file), the service stopps, the ESPs get the EOF signal. 
This could be a solution for single songs, as long as they are not mixed together. It is no solution for podcasts which are up to 2,5 hours. Also, playing from other devices (guest phones..) would be very complicated.

schreibfaul1, 2020-07-19:
There are many ways that clients can become unsynchronized. I am currently unable to replicate the configuration you used. But maybe it helps to change the clock source of the I2S (from PLL_D2_CLK to more accurate APLL_CLK) in "Audio.h"
````
 .use_apll = APLL_ENABLE,
````

see https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#page=306

Insomniac80, 2020-07-23:
Had little time the last days, however, when checking, it became same (maybe even faster) asynchron when using APLL_ENABLE. I will check deeper at the weekend, also i have the guess mp3 vbr makes more problem then mp3 cbr, no matter the bitrate.

Your name lets me assume you are located in a german speaking europe country. If you are interested in replicating the problem, I contact you by private (f.e. youtube) message and you can give me an address I send the stuff you need. Should have everything lying around here (Germany).

timotoots, 2020-08-01:
Just a tip about multi-room audio. There is a nice thing called Snapcast. This would be a fun thing to run on ESP32.

Insomniac80, 2020-08-01:
> Just a tip about multi-room audio. There is a nice thing called Snapcast. This would be a fun thing to run on ESP32.

Reads very interesting, especially
https://github.com/badaix/snapcast/issues/520
I will definitely keep an eye on it. But it seems to be not in a state close to be finished soon. 
So for the moment I just calculate the offset of a  ESP32-node by comparing runtime and uptime at start  of the stream and the actual moment, that´s accurate down to ~15ms, and the value can be shared among the nodes without being affected by the network delay. 
One node then calculates: If two nodes are more then ~50ms asynchronous to each other, I just execute pause/resume twice on all nodes (have to take a look at the source, surely theres a more direct way). This results in a short silence (less then one sec), and they all start from the same position again.
It has to be executed all 10 - 30 minutes, sounds fine so far. I think I will finalize it this weekend  (hopefully^^).
While implementing it, I was able to see all nodes become asnchronous while playing, no idea why. On the lucky side, mostly it is the same amount of ms on all of them, so they stay sync. But it makes me think multirooming using Snapcast will not be an easy thing, as I assume the delay occurs on ESP-side.

schreibfaul1, 2020-08-03:
Hello Insomniac80,
as far as I understand that with Multiroom Audio the playback speed of the clients is achieved by a slight change in the sample rate and thus all clients are kept in sync. Audiocast and Snapcast are two well-known examples of this. Maybe that would be something for you: https://github.com/jorgenkraghjakobsen/snapclient

closed 2020-10-10 by stale[bot]