schreibfaul1/ESP32-audioI2S #52: Bluetooth Speaker with ESP32?

rtek1000, 2020-07-25:
I found a Bluetooth Speaker project with ESP32:

[[#10] DIY DSP Bluetooth Speaker with ESP32](https://www.youtube.com/watch?v=_nCOluMHfGI)

Would it be possible to add this functionality to the project?

rtm-soft, 2020-09-11:
Looking for this too, I want to switch between http, sd and bluetooth streaming on the ESP32-A1S-Audio Kit. I found a repository that will play bluetooth on this hardware in Arduino IDE. I tried to implement the software but because of the 2 different ways this software works it crashes. I am thinking of using RTOS to individually load the 2 different pieces of software separately to makes this work. In http/sd mode I load the audio class. Then close the task and then load the bluetooth task. Will be continued, my C(++) skill are limited so I might not be able to get it to work, we will see. TBC.

Link to Arduino Bluetooth player for ESP32 Audio Kit:
https://github.com/RealCorebb/ESP32-A1s-Audio-Kit

schreibfaul1, 2020-09-11:
I also had the idea to activate BT, not as a sink but as an A2DP source as a replacement for the ADC. Unfortunately, BT-Classic uses a lot of RAM, so that there is not enough left for the decoder.

closed 2020-11-19 by stale[bot]

drtrigon, 2023-09-11:
> I also had the idea to activate BT, not as a sink but as an A2DP source as a replacement for the ADC. Unfortunately, BT-Classic uses a lot of RAM, so that there is not enough left for the decoder.

What hardware are you considering? Does this also hold for M5Core2 ?

schreibfaul1, 2023-09-11:
BT is not part of this library. Due to the required memory and additional computing time, there might be problems in existing projects. But it is possible to use BT. [ESPuino](https://github.com/biologist79/ESPuino) is a project that uses BT classic.

drtrigon, 2023-09-11:
So that proofs that it is indeed possible to use ESP32-audioI2S together with bluetooth (on M5Stack devices).

Thanks for the hint!