schreibfaul1/ESP32-audioI2S #54: Issue with redirects

wga22, 2020-08-07:
Iheart radio streams use a "301 redirect".  I tried replacing the HTTPS Client with the HTTPSRedirect (https://github.com/electronicsguy/ESP8266/tree/master/HTTPSRedirect - works by replacing "client") , and though the redirect now occurs, it still is not loading the stream.  Could someone provide second set of eyes on what might be the problem?

Sample stream
**https://stream.revma.ihrhls.com/zc4422**
---> redirects to ->>
https://n2ba-e2.revma.ihrhls.com/zc4422?rj-ttl=5&rj-tok=AAABc8lJKsoApd8aBgVQsiq9IQ

schreibfaul1, 2020-08-07:
in https://n29a-e2.revma.ihrhls.com/zc4422?rj-ttl=5&rj-tok=AAABc8rcNRcAJttL4fCfRfvxZw the character string from the '&' is cut off because the further information is not required for 'normal' streams. Remove 
````if (host.indexOf ('&')> 0) host = host.substring (0, host.indexOf ('&')); // remove parameter```` 
(line 1202 in Audio.cpp) and the forwarding works. However, AAC + (PS) cannot be played back correctly.

wga22, 2020-08-09:
Removing the code tripping querystring elements, and replacing Client with RedirectClient, fixed for me!  thank you for the assistance, issue can be closed.

feel free to use anything from my version of the code:
https://github.com/wga22/esp8266/blob/master/arduino/mp3Streamer/mp3streamer/mp3streamer.ino

wga22, 2020-08-14:
> However, AAC + (PS) cannot be played back correctly.
Can you elaborate on what you mean by this?  I am able to get the streams to work, but they glitch on ~30-60 second intervals.

schreibfaul1, 2020-08-18:
Hello wga22, forget SBR / PS, IHeart transmits unusually short frames. I changed my input buffer concept. Now there are no more dropouts.

closed 2020-10-28 by stale[bot]