schreibfaul1/ESP32-audioI2S #59: extend pause/resume

Misz0r, 2020-08-29:
Hi,

It's not really an issue of this project. Is it possible to write the audio position to the sd card? If i try to get the audio position by calling audio.getFilePos(); the esp32 crashes. 
what am I doing wrong?

abort() was called at PC 0x400854c9 on core 1
Backtrace: 0x4008c8c8:0x3ffbe660.......................

schreibfaul1, 2020-08-29:
this works for me, log filepos every second

// in global
Ticker ticker;

// in setup
  ticker.attach(1, tcr1s);

void tcr1s(){
    uint32_t pos =audio.getFilePos();
    log_i("pos =%i", pos);
}

Misz0r, 2020-09-01:
you are totally right. As I wrote above it seems to be a problem how i try to call this function. I try to get the FilePos invoked by an hardware interrupt (pressed button). I think this causes my error. Setting the volume is no problem. This is my function
`void playPause_pressed() {  
  static unsigned long last_interrupt_time = 0; //for debouncing
  unsigned long interrupt_time = millis();//for debouncing
  if (interrupt_time - last_interrupt_time > 100) //for debouncing
  {
      Serial.println("Interrupt playPause_pressed!");
      audio.pauseResume();
      audio.getFilePos();
  }
  last_interrupt_time = interrupt_time; //for debouncing
}`

//in setup()
attachInterrupt(digitalPinToInterrupt(btn_playPause.PIN), playPause_pressed, FALLING);

Misz0r, 2020-09-01:
Sorry for bothering you. You did a great job. The only issues are my first steps in hardware programming.

closed 2020-09-01 by Misz0r

schreibfaul1, 2020-09-01:
Interups are not easy
Here is a suggested solution (not tested);
````
// global
DRAM_ATTR bool f_play_pause = false; // set from isr

// setup()
attachInterrupt(digitalPinToInterrupt(btn_playPause.PIN), playPause_pressed, FALLING);

// loop()
if(f_play_pause){
    Serial.println("Interrupt playPause_pressed!");
    audio.pauseResume(); 
    audio.getFilePos();
    ...
    ...
    f_play_pause = false;
}

void IRAM_ATTR playPause_pressed(){
    f_play_pause = true;
}
````