schreibfaul1/ESP32-audioI2S #60: Interrupted sound

buhahapl, 2020-09-02:
Hi,
I was searching for high quality 320kbps stations and i found some with periodically interrupted sound.
I found other 320kbps stations which plays well, so it is not ESP32 performance problem. I checked and it is not PSRAM buffer problem.
I'm wondering why some stations didnt plays correctly?
I loked at ICY HTTP response headers but I didin't found anything special...
Maybe problematic stations have different mpeg format and MP3 decoder cannot handle it?

Do you have any idea what is wrong here?
thanks, wk

Here are examples:

**NOT working properly:**
// 320kbps
http://listen.radiogora.ru:8000/electro320
http://play.organlive.com:7000/320
// 128kbps
http://thecheese.ddns.net:8010/listen.pls

**Working ok:**
//320kbps
https://mediaserv30.live-streams.nl:18086
http://www.partyviberadio.com:8010/listen.pls
http://www.partyviberadio.com:8000/listen.pls

**Not OK station header:**
HTTP/1.0 200 OK
icy-notice1:<BR>This stream requires <a href="http://www.winamp.com">Winamp</a><BR>
icy-notice2:SHOUTcast DNAS/win32 v2.6.0.750<BR>
Accept-Ranges:none
Access-Control-Allow-Origin:*
Cache-Control:no-cache,no-store,must-revalidate,max-age=0
Connection:close
icy-name:Organlive.com
icy-genre:classical
icy-br:320
icy-sr:44100
icy-url:http://www.organlive.com
icy-pub:1
content-type:audio/mpeg
icy-metaint:16384
X-Clacks-Overhead:GNU Terry Pratchett
onNewStation: Organlive.com

**OK station header:**
HTTP/1.0 200 OK
Server: Icecast acc-0.9.10.25 2.4.3-backportCVE-2018-18820
Date: Wed, 02 Sep 2020 10:49:37 GMT
Content-Type: audio/mpeg
Cache-Control: no-cache
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Pragma: no-cache
Access-Control-Allow-Origin: *
icy-br:320
ice-audio-info: channels=2;samplerate=44100;bitrate=320
icy-br:320
icy-name:RadioMonster.FM - Tophits
icy-pub:1
icy-metaint:16000

buhahapl, 2020-09-02:
I'm pretty sure that the problem is in the mp3_decoder class (or its usage?).
I prepare large buffer filled with mp3 data, syncword is found correctly - MP3FindSyncWord() returns zero and MP3Decode() reports no error (returns zero). Most of streams plays correcttly but the ones I have indicated aren't.

ranger2012, 2020-09-03:
i normally ping the site address first , if response time is above 280ms, blurrping is sure to happen specially above 128kbps bitrate...  
pinging   http://thecheese.ddns.net:8010/listen.pls  

Pinging thecheese.ddns.net [114.23.160.66] with 32 bytes of data:
Reply from 114.23.160.66: bytes=32 time=137ms TTL=48
Reply from 114.23.160.66: bytes=32 time=85ms TTL=48
Reply from 114.23.160.66: bytes=32 time=93ms TTL=48
Reply from 114.23.160.66: bytes=32 time=104ms TTL=48

response time is very good, so i have no problem playing this station from my side..

buhahapl, 2020-09-03:
I checked buffer filling and the problem is insuficient data available on WifiCient. There are buffer underruns. The question is why the server not sending enough data?
I can play those stations in Chrome using the same internet provider without any problems...

buhahapl, 2020-09-04:
@ranger2012

Can you run code above on your ESP32 and check the results?
thanks, wk

ranger2012, 2020-09-05:
``hi Buhahapl, Here's the result.. from my side (location is Australia)`

14:56:54.292 -> Connected to stream: http://thecheese.ddns.net:8010/
14:56:54.292 -> 
14:56:54.292 -> 4308, 5744, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 
14:56:55.275 -> --------------------------
14:56:55.316 -> 32 kB/s
14:56:55.316 -> --------------------------
14:56:55.316 -> 5744, 0, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 
14:56:56.308 -> --------------------------
14:56:56.308 -> 30 kB/s
14:56:56.308 -> --------------------------
14:56:56.308 -> 5744, 1436, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 
14:56:57.297 -> --------------------------
14:56:57.297 -> 29 kB/s
14:56:57.297 -> --------------------------
14:56:57.297 -> 5744, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 5744, 
14:56:58.307 -> --------------------------
14:56:58.307 -> 30 kB/s
14:56:58.307 -> --------------------------
14:56:58.307 -> 0, 5744, 5744, 1436, 5744, 0, 5744, 0, 5744, 0, 
14:56:59.322 -> --------------------------
14:56:59.322 -> 30 kB/s
14:56:59.322 -> --------------------------
14:56:59.322 -> 5744, 5744, 0, 5744, 0, 5744, 0, 5744, 5744, 5744, 
14:57:00.339 -> --------------------------
14:57:00.339 -> 32 kB/s
14:57:00.339 -> --------------------------
14:57:00.339 -> 5744, 5744, 5744, 5744, 5744, 5744, 5744, 5744, 5744, 5744, 
14:57:01.366 -> --------------------------
14:57:01.366 -> 35 kB/s
14:57:01.366 -> --------------------------
14:57:01.366 -> 5744, 0, 5744, 1436, 5744, 0, 5744, 0, 5744, 0, 
14:57:02.366 -> --------------------------
14:57:02.366 -> 34 kB/s
14:57:02.366 -> --------------------------
14:57:02.366 -> 5744, 0, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 
14:57:03.365 -> --------------------------
14:57:03.365 -> 33 kB/s
14:57:03.365 -> --------------------------
14:57:03.365 -> 5744, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 5744, 
14:57:04.377 -> --------------------------
14:57:04.377 -> 33 kB/s
14:57:04.377 -> --------------------------
14:57:04.377 -> 0, 5744, 0, 5744, 0, 5744, 0, 5744, 0, 5744, 
14:57:05.365 -> --------------------------
14:57:05.365 -> 33 kB/s
14:57:05.365 -> --------------------------
14:57:05.399 -> 0, 5744, 0, 5744, 0, 5744, 0, 5744, 

Connecting WiFi.
15:00:19.506 -> Connected to stream: http://188.165.192.5:8853/stream
15:00:19.506 -> 
15:00:19.506 -> 5231, 0, 0, 0, 0, 4308, 0, 0, 0, 5744, 
15:00:20.526 -> --------------------------
15:00:20.526 -> 14 kB/s
15:00:20.526 -> --------------------------
15:00:20.526 -> 0, 0, 0, 0, 5744, 0, 0, 0, 0, 5744, 
15:00:21.515 -> --------------------------
15:00:21.515 -> 13 kB/s
15:00:21.515 -> --------------------------
15:00:21.515 -> 0, 0, 0, 0, 5744, 0, 0, 0, 0, 5744, 
15:00:22.504 -> --------------------------
15:00:22.504 -> 12 kB/s
15:00:22.504 -> --------------------------
15:00:22.504 -> 0, 0, 0, 5744, 0, 0, 0, 5744, 0, 0, 
15:00:23.507 -> --------------------------
15:00:23.507 -> 12 kB/s
15:00:23.507 -> --------------------------
15:00:23.507 -> 0, 5744, 0, 0, 0, 5744, 0, 0, 0, 5744, 
15:00:24.519 -> --------------------------
15:00:24.519 -> 13 kB/s
15:00:24.519 -> --------------------------
15:00:24.519 -> 0, 0, 0, 5744, 0, 0, 0, 0, 5744, 0, 
15:00:25.517 -> --------------------------
15:00:25.517 -> 12 kB/s
15:00:25.517 -> --------------------------
15:00:25.517 -> 0, 0, 0, 5744, 0, 0, 0, 0, 5744, 0, 
15:00:26.545 -> --------------------------
15:00:26.545 -> 12 kB/s
15:00:26.545 -> --------------------------
15:00:26.545 -> 0, 0, 0, 5744, 0, 0, 0, 5744, 0, 0, 
15:00:27.542 -> --------------------------
15:00:27.542 -> 12 kB/s
15:00:27.542 -> --------------------------
15:00:27.542 -> 0, 5744, 0, 0, 0, 5744, 0, 0, 0, 5744, 
15:00:28.539 -> --------------------------
15:00:28.539 -> 12 kB/s
15:00:28.539 -> --------------------------
15:00:28.539 -> 0, 0, 0, 5744, 0, 0, 0, 5744, 0, 0, 
15:00:29.536 -> --------------------------
15:00:29.536 -> 12 kB/s
15:00:29.536 -> --------------------------
15:00:29.536 -> 0, 5744, 0, 0, 0, 1436, 4308, 0, 0, 0, 
15:00:30.555 -> --------------------------
15:00:30.555 -> 12 kB/s
15:00:30.555 -> --------------------------
15:00:30.555 -> 5744, 0, 0, 0, 0, 5744, 0, 0, 0, 0, 
15:00:31.534 -> --------------------------
15:00:31.534 -> 12 kB/s
15:00:31.534 -> --------------------------
15:00:31.534 -> 5744, 0, 0, 0, 5744, 0, 0, 0, 5744, 0, 
15:00:32.530 -> --------------------------
15:00:32.569 -> 12 kB/s
15:00:32.569 -> --------------------------
15:00:32.569 -> 0, 0, 5744, 

```

ranger2012, 2020-09-05:
hi Wiktor,
i've posted the result on Github..
you've also visited Earlephilhower  esp8266 github site.. i love his works too,, specially the I2SNODAC  class..  so simply yet.. its loud and clear as well.. i wanted to copy that class ... and see how it works  on Wolle's code,  my favorite music streaming is the calm radio solo piano.. but the metadata is transferred in Chunked transfer encoding.. which wasn't handled correctly.. so there's a lot clicks coming in... so far Wolle's code is Ok for me  at the moment..

cheers,
jose

buhahapl, 2020-09-05:
Hello Ranger,
You have opposite results than mine. The station plays well on my location (Poland) plays incorrect in your location and vice versa ;) Probably the result depends on distance between server and client and routing.
Basing on WiFiClient class i created my own implementation using socket and not using streams but only memory buffer (the type of reading is quite simple - we need to read all incoming data and all additional methods are not necessary).
At the begining the results were similar, buf I modified read(...) method:
int res = recv(sockfd, _buffer, _size , MSG_DONTWAIT);
MSG_DONTWAIT was replaced by MSG_MORE.
Now there are no gaps during reading and for 128kbps stream max speed is 24-27kB/s (of 16kB/s required), so it is OK.

473, 4308, 1436, 4308, 1436, 4308, 1436, 4308, 1436, 4308, 
--------------------------
27 kB/s
--------------------------
1436, 4308, 1436, 4308, 1436, 4308, 1436, 4308, 1436, 4308, 
--------------------------
27 kB/s

But for some problematic 320kbs streams this is not enough.
Also I didn't check the impact of  this change on performance and decoding.

Probably the problem is in ESP32 network libraries.
If I have a time I add new issue for ESP32 project on https://github.com/espressif/arduino-esp32/issues and we'll see what happen...

Some time ago i I see Earlephilhower implementation but it has problems with decoding streams with small metadata intervals.
I don't remember if I checked problematic streams on classes without icy metedata decoding, but probably yes and it also not playing correctly.
There are also FLAC decoder there, which may be changed and used in this project...

thanks, wk

ranger2012, 2020-09-06:
Hi Wiktor, 
would you be able to share you code, using web sockets,, and i will try on my side, if there's no more problem  receiving Calm Radio Solo piano streams.. 

Thanks,
Jose

wiktorbpl, 2020-09-06:
Hello Jose,
I'm affraid I was too fast and optimistic. Changing option used in recv() causes socked waiting for data and only if any arrives, the code can continue. So speed measuring was improper. I checked affected stations and the result was the same :(
I also modify ESP WIFI settings and increase Rx buffers, but it also does nothing.
This is definitely a hardware or arduino esp32 package libraries problem.
I will report this problem on ESP32 github forum.
thanks, wk

rtm-soft, 2020-09-13:
Hi,
I looked at this too and had the same problem, so I played around with the program and I might have solved some issues. I think if you keep polling the wifi client buffer it actually slows down the streaming process. Since the ESP32 has a dual core processor I modified the program with one core dealing with receiving the data and the other core with display tasks. Also I modified the code to show kpbs as this has to be higher then the kpbs of the audio stream ;-) So the trick is to wait with polling until the buffer is about filled and then read the buffer. 

`#include "Arduino.h"
#include "WiFi.h"
#include "WiFiClient.h"
#include "HTTPClient.h"

String ssid =     "SSID";
String password = "password";

void setup()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());
  Serial.print("Connecting WiFi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(1000);
  }
  Serial.println();

  Serial.printf("loop()...running on core: %d\r\n", xPortGetCoreID());
  xTaskCreatePinnedToCore(AudioTask, "Audio Task", 16384, NULL, 2, NULL, 0);
}

int sum = 0;
int samples = 0;
int rate = 0;
int wasRead = 0;
bool bufferfull= false;
const int buffLen = 10*1024;
uint8_t buffer[buffLen];

void loop()
{
  for(int i=0; i<10; i++)
  {
    while (!bufferfull) vTaskDelay(10);
    bufferfull=false;
    if(wasRead < 0)
      wasRead = 0;
    Serial.print(wasRead);Serial.print(", ");
    if(wasRead > 0)
      sum += wasRead;
    wasRead=0;
    samples++;
  }
  rate = ((sum / samples) *10*8) / 1024;
  Serial.println();
  Serial.println("--------------------------");
  Serial.print(rate);
  Serial.println(" kb/s");
  Serial.println("--------------------------");
}

void  AudioTask( void * parameter )  {
  Serial.printf("AudioTask...running on core: %d\r\n", xPortGetCoreID());

  WiFiClient client;
  HTTPClient http;
  // const char *url = "http://188.165.192.5:8853/stream"; // 128kbps good
  const char *url = "http://thecheese.ddns.net:8010/"; // 128kbps bad
  //const char *url = "http://94.130.242.5:8010/stream"; // 320kbps good
  //  const char* url = "http://play.organlive.com:7000/320"; // 320kbps bad
  
  const char *hdr[] = { "icy-metaint", "icy-name", "icy-genre", "icy-br" };
  http.begin(client, url);
  http.addHeader("Icy-MetaData", "1");
  http.collectHeaders( hdr, 4 );
  http.setReuse(true);
  int code = http.GET();
  if (code != HTTP_CODE_OK)
  {
    http.end();
    Serial.println("Not connected to stream!");
  }
  else
  {
    Serial.print("Connected to stream: "); Serial.println(url); Serial.println();
  } 
  while (true) {
    while (wasRead<1) wasRead = client.read(buffer, buffLen);
    bufferfull=true;
    vTaskDelay(200);                    //Time to do background tasks and fill the buffer
  }
}`

wiktorbpl, 2020-09-14:
@rtm-soft

Hi.
I think following:
`while (!bufferfull) vTaskDelay(10);`
in the time measuring loop causes wrong speed calculation.
Leave the code as it is and add calculating rate using time:
```
start = millis();
// the loops
end = millis();
elapsed = end - start;
rate = (sum / (elapsed / 1000)) / 1024; // (in kB/s)
```

I add new issue on arduino ESP32 GitHub:
https://github.com/espressif/arduino-esp32/issues/4337
but there is no answer yet.

thanks, wk

ranger2012, 2020-09-14:
@rtm-soft 

wow, looks interesting, thanks for sharing the code, i think i'll be busy  testing your code on some URLs .. and see if there's improvement.

schreibfaul1, 2020-09-14:
Or the ticker is used as the time base, then the program looks like this:
````
#include "Arduino.h"
#include "WiFi.h"
#include "WiFiClient.h"
#include "HTTPClient.h"
#include "Ticker.h"

Ticker ticker;

String ssid = "SSID";
String password = "password";

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid.c_str(), password.c_str());
    Serial.print("Connecting WiFi");
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(1000);
    }
    Serial.println();
    ticker.attach(1, tckr1s);

    Serial.printf("loop()...running on core: %d\r\n", xPortGetCoreID());
    xTaskCreatePinnedToCore(AudioTask, "Audio Task", 16384, NULL, 2, NULL, 0);
}

int sum = 0;
int samples = 0;
int rate = 0;
int wasRead = 0;
bool bufferfull = false;
const int buffLen = 10 * 1024;
uint8_t buffer[buffLen];

void loop() {
        while(!bufferfull)
            vTaskDelay(10);
        bufferfull = false;
        if(wasRead < 0)
            wasRead = 0;
        Serial.print(wasRead);
        Serial.print(", ");
        if(wasRead > 0)
            sum += wasRead;
        wasRead = 0;
}

void AudioTask(void *parameter) {
    Serial.printf("AudioTask...running on core: %d\r\n", xPortGetCoreID());

    WiFiClient client;
    HTTPClient http;
// const char *url = "http://188.165.192.5:8853/stream"; // 128kbps good
    const char *url = "http://thecheese.ddns.net:8010/"; // 128kbps bad
//const char url = "http://94.130.242.5:8010/stream"; // 320kbps good
// const char *url = "http://play.organlive.com:7000/320"; // 320kbps bad

    const char *hdr[] = { "icy-metaint", "icy-name", "icy-genre", "icy-br" };
    http.begin(client, url);
    http.addHeader("Icy-MetaData", "1");
    http.collectHeaders(hdr, 4);
    http.setReuse(true);
    int code = http.GET();
    if(code != HTTP_CODE_OK) {
        http.end();
        Serial.println("Not connected to stream!");
    } else {
        Serial.print("Connected to stream: ");
        Serial.println(url);
        Serial.println();
    }
    while(true) {
        while(wasRead < 1)
            wasRead = client.read(buffer, buffLen);
        bufferfull = true;
        vTaskDelay(200); //Time to do background tasks and fill the buffer
    }
}

void tckr1s(){ // called  every second
    rate = sum / 1024;
    Serial.println();
    Serial.println("--------------------------");
    Serial.print(rate);
    Serial.println(" kb/s");
    Serial.println("--------------------------");
    sum = 0;
}
````

wiktorbpl, 2020-09-15:
Not working either. Rate is still 12-14kB per second, sometimes 15kB/s and no more.

ranger2012, 2020-09-16:
sounds like its a limitation of ESP32.. 240mhz  compared to 1.2Ghz  of  phones, tablets and pcs

ranger2012, 2020-09-17:
Wolle,
how do I integrate your Audio Class  in the code above, 12 to 14kb is good enough for me, I'm happy with 128 kb  data stream..

schreibfaul1, 2020-09-17:
Hello ranger2012,
the example shows that there is no speed advantage due to an additional task. The limitation is somewhere in the ESP32 core.

closed 2020-11-24 by stale[bot]