schreibfaul1/ESP32-audioI2S #64: play a specified file on command

Samwarez, 2020-10-09:
I have an SD card filled with small .wav files and I am trying to find a way to play specific ones on command. The current examples queue up a single file to be played in the setup() then plays it when the board starts.

What I am looking to do is play a list of files from a function so say I press a button the triggers the function and plays a file, press a different button and a different file plays. I will also need to string a few files together to play one after the other.

Here is what I have so far:
```
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26
#define buttonPin      2
#define ledPin         0

int buttonState = 0; 
    
Audio audio;
void setup() {
    pinMode(SD_CS, OUTPUT);
    pinMode(ledPin, OUTPUT);     
    pinMode(buttonPin, INPUT);
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    SD.begin(SD_CS);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(10); // 0...21
    audio.connecttoFS(SD, "100.wav");
}

void loop()
{
    buttonState = digitalRead(buttonPin);

    if (buttonState == HIGH) {
    digitalWrite(ledPin, HIGH);
    playsound();
  } else {
    digitalWrite(ledPin, LOW);
  }
}

void playsound()
{
  audio.loop();
}

```
This works, but only once before I need to reset the board.

```
void setup() {
    pinMode(SD_CS, OUTPUT);
    pinMode(ledPin, OUTPUT);     
    pinMode(buttonPin, INPUT);
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    SD.begin(SD_CS);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(10); // 0...21
    // audio.connecttoFS(SD, "100.wav"); <---Removed
}

void loop()
{
    buttonState = digitalRead(buttonPin);

    if (buttonState == HIGH) {
    digitalWrite(ledPin, HIGH);
    playsound();
  } else {
    digitalWrite(ledPin, LOW);
  }
}

void playsound()
{
  audio.connecttoFS(SD, "100.wav");
  audio.loop();
}
```
This does not, nothing plays.

So the main question is how do I get that connecttoFS bit to work outside of the main setup funtion?

Eventually I want something akin to this:

void playsound (string file) {
  audio.connecttoFS(SD, file);
  audio.loop();
}

Then somewhere else:
playsound ("Hello.wav");
playsound ("World.wav");
if buttonPressed=1 {
playsound ("Foo.wav");
} elseif buttonPressed=2 {
playsound ("bar.wav");
}

Any ideas on how to get this to work with this library?

nodoubtman, 2020-11-27:
Hey,
You must put audio.loop in the 
void loop()

KevScott, 2020-12-10:
Your calls to the `audio.loop()` method are fine and you have them called from the `loop() `function which is the correct way to do it. Each time you call the `audio.loop()` method it takes some more of the wav file from the SD card and sends it to the I2S interface. This method is also aware of where it is in the file and when it gets to the end of the wav file it just stops sending any further data over the I2S bus.

Where the problem lies is with your `audio.connecttoFS(...)` method. If you follow the source code for this, you will see that each time it is called, all of the buffers are reset. What you need to do is detect when a button has gone from LOW to HIGH (i.e. **the transition**) and on the one loop iteration where you detect that, you call `audio.connecttoFS(...)` with the file name you want to play. That will reset the buffers (which is what you want in this case) and subsequent calls in `loop()` to the `audio.loop()` method will result in the specified wave file being played.

So maybe have something like this in your `loop()`, note I have introduced a new `bool `called `prevButtonState` which you should initialise to the same as `buttonState` in the `setup()` function :
```
void loop()
{
  buttonState = digitalRead(buttonPin);

  if (buttonState == HIGH) {
    digitalWrite(ledPin, HIGH);
    // is this the transition event, if so reset the buffers and play the sound
    if (prevButtonState == LOW) audio.connecttoFS(SD, "100.wav");
  } else {
    digitalWrite(ledPin, LOW);
  }
  // note that once we have called connecttoFS() we should keep calling the 
  // loop method to complete the playing of it, irrespective of the button state.
  audio.loop();
  prevButtonState = buttonState;
}
```
I don't have access to hardware to test the above code but I would expect it to work. I use the following code to repetitively play the first six seconds of a thirty second file and know it works fine :

```
void loop() {

  if ((millis() - myTime) > 6000) {
    // 6 seconds have passed so start again
    myTime = millis();
    audio.connecttoFS(SD, "/Test.mp3");
  } 
  audio.loop();
}
```

closed 2021-02-21 by stale[bot]