schreibfaul1/ESP32-audioI2S #67: Cant stream from secure servers

dgrogandesign, 2020-10-15:
HI there. I have managed to stream from the included servers and some other http servers but whenever i try to stream from an https server i get the error audio/mp3 seen, "info unknown audio format "

BCsabaEngine, 2020-11-15:
I can download/play from https server without problem. Now i stored MP3 as static file, so Apache send proprietrary mime info. Can you send a sample https URL in private?

closed 2020-11-17 by schreibfaul1

schreibfaul1, 2020-11-17:
mime type audio/mp3 is now supported

BCsabaEngine, 2020-11-17:
Thank you on behalf of dgrogandesign!

dgrogandesign, 2020-11-23:
> I can download/play from https server without problem. Now i stored MP3 as static file, so Apache send proprietrary mime info. Can you send a sample https URL in private?

Hi, sorry for late replys. I have actually got it working with https now, to be specific, the problem i was having was trying to access files stored on an amazon AWS server. I tried storing it on a google cloud server and it works. problem is that i would rather use the amazon server...

schreibfaul1, 2020-11-23:
How does the access to AWS work, maybe the authentication fails? What is logged in audio_info ()?

commit c0fe9a71 (2020-11-17): fix #67