schreibfaul1/ESP32-audioI2S #70: Defaults for m_BCLK,m_LRC and m_DOUT interfere with other things

Paulham, 2020-10-22:
Hi
I was working with your **great** library, and with CelliesProjects/wm8978-esp32 on a M5Stack Core Grey with Node attached (WM8978). (I'm new to M5Stacks).
It created a high pitched noise which really warmed up the speaker in the Grey.
I think the issue is that the constructor Audio::Audio defaults to pin 25 for m_LRC  - and pin 25 on the Core is the speaker. If I change these in your code in Audio::Audio it doesn't make the noise. 
It also seems to affect the LCD, perhaps because your default for m_DOUT is 27 which is used for the LCD.
Not sure if this is a big issue, but it took me a few days to work out to change Audio::Audio - and maybe it might impact others?
Thanks for making this library available!

CelliesProjects, 2020-10-23:
Thanks for the tip. I had some trouble setting up M5 Node, but didn't get around to investigate.

Paulham, 2020-10-23:
Hi 
I think I tried that but my C++ isn't good enough - it seemed not to find the longer constructor when doing 'Audio audio'. 
So I created two and put the bulk of the code into a new function.
I'll try your approach tonight.
Thanks
Paul

`
Audio::Audio(uint8_t bclk, uint8_t lrc, uint8_t dout, uint8_t din) {
    audiosetup(bclk, lrc, dout, din);
}

Audio::Audio() {
    audiosetup(5, 13, 2, 34);
}

void Audio::audiosetup(uint8_t bclk, uint8_t lrc, uint8_t dout, uint8_t din)
{
    //i2s configuration
....
    i2s_driver_install((i2s_port_t)m_i2s_num, &m_i2s_config, 0, NULL);

    m_BCLK   = bclk;                        // Bit Clock
    m_LRC    = lrc;                         // Left/Right Clock
    m_DOUT   = dout;                        // Data Out
    m_DIN    = din;                         // Data In

    setPinout(m_BCLK, m_LRC, m_DOUT, m_DIN);
......
 }    
}
`

CelliesProjects, 2020-10-23:
To keep the 'old' ctor working you give the definition (.h) a default value to fall back on if no values are passed in.

CelliesProjects, 2020-10-23:
Should be fixed by #71.

Paulham, 2020-10-24:
It does fix the LCD issue too. Thank you.
I was also wondering why the pins are set in the constructor at all - is that necessary?
I am guessing the issue was that the clock was started then routed to gpio 25 - the speaker - and perhaps that could also be fixed by not setting the pins until the call to setpinout as in your example code M5Stack.ino?
I tried looking at setpinout to see what it did but it seemed complex.
If not then you could remove the setpinout call in M5Stack.ino.

Paulham, 2020-10-24:
What is still strange is why the setpinout in the example code didn't set the right pins and set LCD and speaker to be ok. 
With the new ctor (which works great, thank you!) if you then issue 
```
 audio.setPinout(25, 26, 27, 28);  //clearly rubbish 'pins' but the 25 should cause a bad noise and the rest should stop it working
```
it does nothing, so it works fine.  Which seems wrong because it should redirect the clock to pin 25 and make a bad noise.

This is in the same location as in your original code, but has no effect.

So it seems either setpintout does nothing at this point, or the constructor is run *after* it an sets the pins properly.

I did read this post which implies constructors can be run in non-defined places?
https://forum.arduino.cc/index.php?topic=45794.0

Even if I put the setpinout in the 
```
void loop() {
    audio.setPinout(25, 26, 27, 28);
    audio.loop();
}
```
it doesn't break the new ctor!

CelliesProjects, 2020-10-24:
https://github.com/schreibfaul1/ESP32-audioI2S/blob/b9f787eed85be1640b1eb1fc3f217b74d66810fb/src/Audio.cpp#L2012-L2026
The call to `i2s_set_pin()` returns a error code which isn't checked in ESP32-AudioI2S. It just return a `true`.

See:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv411i2s_set_pin10i2s_port_tPK16i2s_pin_config_t

I didn't test it, but the call to `i2s_set_pin` probably returns a `false` after some checking and does nothing.

CelliesProjects, 2020-10-25:
If #72 will be merged 'setPinout' will return the result of 'is2_set_pin' as a boolean.

Paulham, 2020-10-29:
Hi
Thanks for all your fixes!
I think the return of a bool doesn’t help - I don’t think the ESP library actually returns anything but ESP_OK - when I call setPins it js always ok.
So whilst it works now I can’t figure out why it didn’t before - it is like setPins only works once after a restart and a second call looks ok but has no effect. 
Perhaps it is the hardware somehow.
Thanks for your help - and now it works I think I’ll stop worrying about why it didn’t!

closed 2021-01-04 by stale[bot]