schreibfaul1/ESP32-audioI2S #75: Usage of the library with a CS4344

majuss, 2020-11-24:
Hi guys,

thank you for the great project! I'am trying to use the audio code with the README example in combination with a CS4344. I initialised the master clock and played around with the sampling rate and bits per sample. But no matter what I try, the stream interrupts after a few seconds and I hear squeaky noises.
Here is the datasheet of the DAC: https://www.mouser.de/datasheet/2/76/CS4344-45-48_F2-1141644.pdf
I think page 12 is maybe from bigger interest because I think some clock values diverge.

majuss, 2020-11-27:
Update: I was too stupid :D forgot to connect the BCLK... So onto the next challenge: the tlv320dac3101 :)

closed 2020-11-27 by majuss

715294454, 2022-12-21:
hello majuss,i am trying to use esp32 playing audio with CS4344,but i dont know how to connect esp32 with CS4344,especially the mclk IO. can you tell me the example of connection and the arduino code? thx ^_^