schreibfaul1/ESP32-audioI2S #78: No sound plays from TTS example?

NonaSuomy, 2020-12-08:
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[I][RadioLEDSs_wm8978.ino:90] setup(): Connected. Starting MP3...
info        PSRAM found, inputBufferSize = 298399 bytes
info        buffers freed, free Heap: 173868 bytes
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
info        SSL has been established, free Heap: 127716 bytes

Nothing plays (works fine from SD or Streaming Music station).

Code

```
#include <WM8978.h> // https://github.com/CelliesProjects/wm8978-esp32
#include <Audio.h>  // https://github.com/schreibfaul1/ESP32-audioI2S
#include <SPI.h>
#include <SD.h>
#include <FS.h>

#ifdef __AVR__
  #include <avr/power.h>
#endif

// TTGO T-Audio 1.6 Schematic
// https://github.com/LilyGO/TTGO-TAudio/blob/master/schematic.pdf

// TTGO T-Audio 1.6 button pin setup.
#define Button2       39
#define Button3       36
#define Button4       34

// TTGO T-Audio 1.6 SD-Card SPI pin setup.
#define SD_CS         13
#define SPI_MOSI      15
#define SPI_MISO      02
#define SPI_SCK       14

// TTGO T-Audio 1.6 RGB LED pin setup.
#define RGB_PIN       22
#define NUM_LEDS      19
#define BRIGHTNESS    50

// TTGO T-Audio 1.6 WM8978 & MPU9250 I2C pins setup.
#define I2C_SDA       19
#define I2C_SCL       18

// TTGO T-Audio 1.6 I2S pins setup.
#define I2S_BCK       33
#define I2S_WS        25
#define I2S_DOUT      26
#define I2S_DIN       27

// TTGO T-Audio 1.6 WM8978 MCLK gpio number.
#define I2S_MCLKPIN    0

Audio audio(I2S_BCK, I2S_WS, I2S_DOUT);
WM8978 dac;

void setup() {
  Serial.begin(115200);
  pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SD.begin(SD_CS);

  /* Setup wm8978 I2C interface */
  if (!dac.begin(I2C_SDA, I2C_SCL)) {
    ESP_LOGE(TAG, "Error setting up dac. System halted");
    while (1) delay(100);
  }

  /* Select I2S MCLK pin */
  audio.i2s_mclk_pin_select(I2S_MCLKPIN);

  WiFi.begin("SSID", "WiFIPassword");
  while (!WiFi.isConnected()) {
    delay(10);
  }
  ESP_LOGI(TAG, "Connected. Starting MP3...");
//    audio.connecttoFS(SD, "/320k_test.mp3");
//    audio.connecttoFS(SD, "test.wav");
//    delay(10000);
//    audio.connecttohost("http://www.wdr.de/wdrlive/media/einslive.m3u");
//    audio.connecttohost("http://macslons-irish-pub-radio.com/media.asx");
//    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.aac"); //  128k aac
//    audio.connecttohost("http://mp3.ffh.de/radioffh/hqlivestream.mp3"); //  128k mp3
//    audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");
//    audio.connecttohost("http://media.ndr.de/download/podcasts/podcast4161/AU-20190404-0844-1700.mp3"); // podcast
audio.connecttospeech("Hello this is a test of text from this module.", "en");

  dac.setSPKvol(40); /* max 63 */
  dac.setHPvol(52, 52);
}

void loop() {
  audio.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}

```

BCsabaEngine, 2020-12-08:
Did you try the translate.google.com direct with your sample text? Is this not too long?

NonaSuomy, 2020-12-08:
Seems to work ok directly from the site but that I’m sure is more of a perfect scenario for it to work as chrome is meant to work with it.

schreibfaul1, 2020-12-08:
In November 2020, GoogleTTS changed its access modalities. A TKK token is no longer required.

closed 2020-12-08 by schreibfaul1