schreibfaul1/ESP32-audioI2S #79: How do you play a wav file then play a radio station?

NonaSuomy, 2020-12-08:
It seems to not play the wav and then plays the station instead. On the console it says that it started the wav before that.

```
audio.connecttoFS(SD, "test.wav");
audio.connecttohost("http://macslons-irish-pub-radio.com/media.asx");

[I][RadioLEDSs_wm8978.ino:180] setup(): Connected. Starting audio...
[I][RadioLEDSs_wm8978.ino:191] setup(): Max stations: 9
info        PSRAM found, inputBufferSize = 298399 bytes
info        buffers freed, free Heap: 172944 bytes
info        Reading file: "test.wav"
info        FormatCode=1
info        Channel=2
info        SampleRate=44100
info        DataRate=176400
info        DataBlockSize=4
info        BitsPerSample=16
info        BitRate=1411200
info        DataLength=278592
info        buffers freed, free Heap: 172684 bytes
info        Connect to new host: "www.antenne.de/webradio/antenne.m3u"
info        Playlist request, entry 1
info        Connect to "www.antenne.de" on port 80, extension "/webradio/antenne.m3u"
info        Connected to server
Pressed 3 - Volume Down: 45
info        Read from playlist
info        Playlistheader: HTTP/1.1 200 OK
info        Playlistheader: date: Tue, 08 Dec 2020 11:50:23 GMT
info        Playlistheader: server: Apache
info        Playlistheader: last-modified: Tue, 04 Feb 2020 14:01:25 GMT
info        Playlistheader: accept-ranges: bytes
info        Playlistheader: content-length: 36
info        Playlistheader: cache-control: max-age=3600
info        Playlistheader: expires: Tue, 08 Dec 2020 12:50:23 GMT
info        Playlistheader: content-type: audio/x-mpegurl
info        Playlistheader: connection: close
info        Playlistheader: 
info        Switch to PLAYLISTDATA
info        Playlistdata: http://stream.antenne.de/antenne
info        Entry 1 in playlist found: http://stream.antenne.de/antenne
info        buffers freed, free Heap: 171496 bytes
info        Connect to new host: "stream.antenne.de/antenne"
info        Connect to "stream.antenne.de" on port 80, extension "/antenne"
info        Connected to server
info        Access-Control-Allow-Origin: *
info        Cache-Control: no-cache
info        redirect to new host "http://s10-webradio.antenne.de/antenne"
info        buffers freed, free Heap: 171268 bytes
info        Connect to new host: "http://s10-webradio.antenne.de/antenne"
info        Connect to "s10-webradio.antenne.de" on port 80, extension "/antenne"
info        Connected to server
info        Access-Control-Allow-Origin: *
info        Cache-Control: no-cache
info        Connection: close
info        audio/mpeg seen.
info        format is mp3
info        MP3Decoder has been initialized, free Heap: 141108 bytes
info        Expires: Mon, 26 Jul 1997 05:00:00 GMT
info        Pragma: no-cache
info        Server: streaMonkey streaming Server Native
info        Sessionid: 5fcf68800af06823edb8d795
info        Set-Cookie: streamingsession=5fcf68800af06823edb8d795; Path=/; Domain=antenne.de; Max-Age=9999999999
info        icy-audio-info: ice-channels=2;ice-samplerate=44100;ice-bitrate=128
bitrate     128000
info        icy-description: ANTENNE BAYERN
info        icy-genre: Pop
info        icy-name: ANTENNE BAYERN
station     ANTENNE BAYERN
info        icy-pub: 1
info        icy-url: http://www.antenne.de
icyurl      http://www.antenne.de
info        Switch to DATA, bitrate is 128000, metaint is 16000
lasthost    http://s10-webradio.antenne.de/antenne
info        inputbuffer is being filled
streamtitle Mariah Carey - All I want for christmas is you
info        StreamTitle="Mariah Carey - All I want for christmas is you"

```

BCsabaEngine, 2020-12-08:
You should wait for end of wav/mp3. You can use the info event about it, call audio.connecttohost when this event arrives. See audio.cpp 1045 line.

NonaSuomy, 2020-12-11:
Are you talking about this line?

https://github.com/schreibfaul1/ESP32-audioI2S/blob/653da3a8752bbef6d08aaac9b4d9099fbee0ea51/src/Audio.cpp#L1015

I can't seem to trigger it without crashing.

Any chance there's some sample code on how to play a local file then play a stream without crashing?

CelliesProjects, 2020-12-11:
wait for `!audio.isRunning()` before starting a new song.

NonaSuomy, 2020-12-30:
Thank you @CelliesProjects worked like a charm.

`
boolean  f_firstboot=false;     // First boot of device for playing startup sound.
boolean  f_secondboot=false;    // Second boot of device for playing last radio station.

void setup(){
  f_firstboot=true; // Play startup sound
}
void loop(){
    if(f_firstboot==true){
      audio.connecttoSD("/test.wav");
      audio.loop();
      f_firstboot=false;
      f_mp3eof=false;
      f_secondboot=true;
    } 
    else if (f_secondboot==true){
      if (!audio.isRunning()){
        audio.connecttohost(setStation(pref.getUInt("station"))); // Last used station
        audio.loop();
        //audio.printDetails();
        if(_station=="") showStation(); // If station gives no icy-name display _stationname
        f_secondboot=false;
      }
    }
}
`

closed 2020-12-30 by NonaSuomy