schreibfaul1/ESP32-audioI2S #81: Clicking noise at the end of playback

fri-sch, 2020-12-12:
Thanks for the library. I use it to play some *.wav files trough a MAX98357A. It works pretty well, but I get a loud click at the end of the playback when the track is finished. Any ideas how to prevent this?

mwalling, 2020-12-17:
I don't normally like to "me too" posts, but this bothered me. I thought it was my bad Audition skills, not cropping on a zero crossing for example, as I tried to reproduce with a file that I didn't export from Audition (I think it was a Windows built in), and it worked without issue.

schreibfaul1, 2020-12-17:
Could it be that an older Espressif release is being used and m_i2s_config.tx_desc_auto_clear has no effect?

fri-sch, 2020-12-17:
> Could it be that an older Espressif release is being used and m_i2s_config.tx_desc_auto_clear has no effect?

I'm using PlatformIO and the [v2.0.0 release](https://github.com/platformio/platform-espressif32/releases/tag/v2.0.0) that seems to use ESP-IDF 4.0.1. I will upgrade to v2.1.0 and see if that will change anything.

For the moment I can tell that the following change seems to "solve" the issue for me without any noticeable downsides:

```
diff --git a/src/Audio.cpp b/src/Audio.cpp
index 76b291f..c6be1b5 100644
--- a/src/Audio.cpp
+++ b/src/Audio.cpp
@@ -981,7 +981,7 @@ void Audio::processLocalFile()
         if(bytesCanBeRead > 1600) bytesCanBeRead = 1600;
         if(bytesCanBeRead == 1600){ // mp3 or aac frame complete?
             if(!m_f_stream){
-                if(!playI2Sremains()) return; // release the thread, continue on the next pass
+                //if(!playI2Sremains()) return; // release the thread, continue on the next pass^M
                 m_f_stream = true;
                 if(audio_info) audio_info("stream ready");
             }
@@ -1004,7 +1004,7 @@ void Audio::processLocalFile()
                 lastChunk = true;
                 return; // release the thread, continue on the next pass
             }
-            if(!playI2Sremains()) return;
+            //if(!playI2Sremains()) return;^M
             stopSong();
             m_f_stream=false;
             m_f_localfile=false;
```

fri-sch, 2020-12-19:
> For the moment I can tell that the following change seems to "solve" the issue for me without any noticeable downsides

Correction: It does have downsides, as some clips are now cut off at the end. I guess that's what `playI2Sremains()` actually should prevent (see #44).

And switching to PlatformIO platform-espressif 2.1.0 didn't help with the clicking noise issue.

fri-sch, 2020-12-19:
> Could it be that an older Espressif release is being used and m_i2s_config.tx_desc_auto_clear has no effect?

If you are talking about https://github.com/espressif/esp-idf/commit/765bf674d0d54b2b871bb2d1e41160bc4025da1a, this is included since esp-idf 3.3, so it shouldn't be a problem.

schreibfaul1, 2020-12-19:
Hello fri-sch,
I use Eclipse as IDE and the current ESP32-Arduino version 1.5.0. The problem doesn't have to have anything to do with tx_desc_auto_clear, that's just a guess. PlayI2Sremains () should transfer the remainder from the memory to I2S after the end of the file has been recognized, so that the song is not ended prematurely.

marc-gist, 2020-12-22:
I am actually getting a clicking at the start of playback :(

closed 2021-03-05 by stale[bot]