schreibfaul1/ESP32-audioI2S #82: DNS-Error

biologist79, 2020-12-14:
Hi Wolle,

maybe you still remember me; I'm using your lib for my Tonuino (https://github.com/biologist79/Tonuino-ESP32-I2S). However, a user of my software tried to stream from http://streamtdy.ir-media-tec.com/weihnachtslieder/mp3-128/web/play.mp3 but this ended up in a DNS-problem. I had the same problem; even just with only your example-sketch.
However, another user was able to get this running with an older version of your software (oct 2019) and indeed, I can confirm this.

So, somehow, it seems as a bug was introduced meanwhile.

`Free heap: 114832
info        : PSRAM not found, inputBufferSize = 6399 bytes
info        : buffers freed, free Heap: 107076 bytes
info        : Connect to new host: "http://streamtdy.ir-media-tec.com/weihnachtslieder/mp3-128/web/play.mp3"
info        : Connect to "streamtdy.ir-media-tec.com" on port 80, extension "/weihnachtslieder/mp3-128/web/play.mp3"
info        : Connected to server
info        : Access-Control-Allow-Origin: *
info        : Cache-Control: no-cache, must-revalidate
info        : audio/mpeg seen.
info        : format is mp3
info        : MP3Decoder has been initialized, free Heap: 82880 bytes
info        : Expires: Mon, 14 Dec 2020 00:16:06 GMT
info        : redirect to new host "http://irmedia.streamabc.net/irm-rtweihnachtslieder-mp3-128-1989390?sABC=5sq6nrp6%231%23cynl.zc3%23jro&=&amsparams=playerid:web;skey:1607904966"
info        : buffers freed, free Heap: 105724 bytes
info        : Connect to new host: "http://irmedia.streamabc.net/irm-rtweihnachtslieder-mp3-128-1989390?sABC=5sq6nrp6%231%23cynl.zc3%23jro&=&amsparams=playerid:web;skey:1607904966"
info        : Connect to "irmedia.streamabc.net/irm-rtweihnachtslieder-mp3-128-1989390?sABC=5sq6nrp6%231%23cynl.zc3%23jro&=&amsparams=playerid" on port 0, extension "/irm-rtweihnachtslieder-mp3-128-1
Batteriespannung niedrig: (0.40 V)
Aktuelle Batteriespannung: 0.40 V
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for irmedia.streamabc.net/irm-rtweihnachtslieder-mp3-128-1989390?sABC=5sq6nrp6%231%23cynl.zc3%23jro&=&amsparams=playerid
info        : Request irmedia.streamabc.net/irm-rtweihnachtslieder-mp3-128-1989390?sABC=5sq6nrp6%231%23cynl.zc3%23jro&=&amsparams=playerid:web;skey:1607904966 failed!
station     : 
streamtitle : `

schreibfaul1, 2020-12-14:
Hi biologist79,
I know the Tonuino from https://www.mikrocontroller.net/topic/474383#new, a great project for children!
I was not aware of the problem at all and had never occurred to me. Thanks for the hint. I hope everything works now
Best regards, Wolle

biologist79, 2020-12-14:
Wow! Thank you for the quick fix and at all for providing that library. Really appreciate that!
Works perfect now.

Greets
Torsten.

closed 2020-12-14 by biologist79

biologist79, 2020-12-14:
Hmm, sorry to bother you again. Seems as this introduced a new problem. The radio-station posted runs fine, but now this one doesn't work (but was ok before the fix):

http://stream.friskyradio.com:9000/frisky_mp3_h

info        : Connect to "stream.friskyradio.com:9000" on port 80, extension "/frisky_mp3_h"
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for stream.friskyradio.com:9000
info        : Request stream.friskyradio.com:9000/frisky_mp3_h failed!

Same for http://stream.bassdrive.uk:8200
info        : Connect to "" on port 80, extension "/"
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for

schreibfaul1, 2020-12-14:
You're right, I made a mistake myself. The port number is not recognized correctly.
I think it should work now

biologist79, 2020-12-14:
Great!