schreibfaul1/ESP32-audioI2S #83: MP3s with an image reboot the esp32.

CelliesProjects, 2020-12-15:
Hi Wolle.

Another issue. 8)
The esp32 reboots every time on certain files. These files play fine on other players. 

The trace is not very helpful to me:
```
Backtrace: 0x400891c8:0x3ffd82f0 0x40089445:0x3ffd8310 0x400914e7:0x3ffd8330 0x40090f16:0x3ffd8350 0x4008393d:0x3ffd8370 0x4008396e:0x3ffd8390 0x40083275:0x3ffd83b0 0x4000beaf:0x3ffd83d0 0x400818d9:0x3ffd83f0 0x4008b4fd:0x3ffd8410 0x4008d3e1:0x3ffd8450 0x40171d3d:0x3ffd84d0 0x4008e35d:0x3ffd8510

Decoding stack results
0x400891c8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40089445: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400914e7: multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_platform.h line 54
0x40090f16: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 191
0x4008393d: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 111
0x4008396e: heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 140
0x40083275: _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 37
0x400818d9: wifi_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_adapter.c line 68
0x4008e35d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

The sketch below tries to play a file with the issue.

```
#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

Audio audio;
WiFiMulti wifiMulti;

String ssid =     "xxx";
String password = "xxx";

void setup() {
    pinMode(SD_CS, OUTPUT);      digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    SPI.setFrequency(1000000);
    Serial.begin(115200);
    SD.begin(SD_CS);
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
        WiFi.disconnect(true);
        wifiMulti.run();
      }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(12); // 0...21

    //audio.connecttoFS(SD, "/320k_test.mp3");
    audio.connecttohost("http://wasietsmet.nl/test_img.mp3");
    //audio.connecttohost("dg-ais-eco-http-fra-eco-cdn.cast.addradio.de/hellwegradio/west/mp3/high");
    //audio.connecttohost("http://macslons-irish-pub-radio.com/media.asx");
    //audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");
}

void loop()
{
    audio.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}
```

schreibfaul1, 2020-12-15:
Hello Cellie,
This is a rare case where ID3 metadata is sent. The mp3 decoder cannot process an image. That would have to be removed from the stream beforehand. I have no idea how to do it yet, let's see ...

schreibfaul1, 2020-12-15:
@Cellie
These are the ID3 data: 
[image]
I determined the length to completely skip the metadata block. Maybe I can read out the data in a later version.
This is what the log now looks like:
[image]
Best regards, Wolle

CelliesProjects, 2020-12-15:
I get the same back trace. 8(

```
info        PSRAM not found, inputBufferSize = 6399 bytes
info        buffers freed, free Heap: 171284 bytes
info        Connect to new host: "http://wasietsmet.nl/test_img.mp3"
info        Connect to "wasietsmet.nl" on port 80, extension "/test_img.mp3"
info        Connected to server
info        accept-ranges: bytes
info        Content-Length: 9768372
info        audio/mpeg seen.
info        format is mp3
info        MP3Decoder has been initialized, free Heap: 140216 bytes
info        last-modified: Tue, 15 Dec 2020 14:31:59 GMT
info        x-cache: yes
info        server: CLB/No-Cache
info        x-clb-cache: No
info        x-clb-hits: 0
info        connection: close
station     
bitrate     
info        Switch to SWM, bitrate is 0, metaint is 0
lasthost    http://wasietsmet.nl/test_img.mp3
info        stream ready
info        buffer filled in 4485 ms
info        ID3 Metadata recognized, skip 65313 bytes
info        slow stream, dropouts are possible
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400914d3  PS      : 0x00060a33  A0      : 0x80090f19  A1      : 0x3ffd8300  
A2      : 0x3ffe0440  A3      : 0x0000066c  A4      : 0x3ffe28a8  A5      : 0x3ffc9130  
A6      : 0x30000000  A7      : 0x3ffd8f60  A8      : 0x00000001  A9      : 0xe7404ae5  
A10     : 0x3ffe28a8  A11     : 0x00000001  A12     : 0x3ffe28ac  A13     : 0x3ffe08c0  
A14     : 0x3fd1ca70  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xe7404ae5  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400914d3:0x3ffd8300 0x40090f16:0x3ffd8320 0x4008393d:0x3ffd8340 0x4008396e:0x3ffd8360 0x40083275:0x3ffd8380 0x4000beaf:0x3ffd83a0 0x400818d9:0x3ffd83c0 0x4008b4fd:0x3ffd83e0 0x4008d079:0x3ffd8420 0x40171e5d:0x3ffd84a0 0x4008e35d:0x3ffd84e0

```
I tried this on several boards/codecs to rule out faulty hardware.

Back trace decoder:
```
PC: 0x400914d3: multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c line 131
EXCVADDR: 0xe7404ae5

Decoding stack results
0x400914d3: multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c line 131
0x40090f16: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 191
0x4008393d: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 111
0x4008396e: heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 140
0x40083275: _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 37
0x400818d9: wifi_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_adapter.c line 68
0x4008e35d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
 I will re-clone your repo, see if there is something wonky with git.

schreibfaul1, 2020-12-15:
You're right, my first version only worked with PSRAM enabled. Please test again

CelliesProjects, 2020-12-15:
Working perfectly again. 

It actually was a major issue in a project of myself so thanks once again for solving 2 issues!

closed 2020-12-15 by CelliesProjects