schreibfaul1/ESP32-audioI2S #84: Bluetooth audio sink instead of I2S output

mtonnie, 2020-12-18:
Is it possible to use a Bluetooth audio sink to play the music on an Bluetooth speaker?

schreibfaul1, 2020-12-18:
To the best of my knowledge, BT speakers expect a sampling rate of 44100 samples /sec. It would work for radio stations with this sample rate, for all others it would have to be sampled up / down. That surely exceeds the capabilities of the ESP32. But maybe someone knows a BT code to test.

biologist79, 2020-12-21:
I was able to use an audio sink using this lib: https://github.com/pschatzmann/ESP32-A2DP.git

schreibfaul1, 2020-12-21:
Hello biologist,
I think mtonnie means with 'Bluetooth audio sink instead of I2S output' that I2S (including an external DAC) will be replaced by a BT speaker. It's not easy to do.

mtonnie, 2020-12-22:
> 
> 
> Hello biologist,
> I think mtonnie means with 'Bluetooth audio sink instead of I2S output' that I2S (including an external DAC) will be replaced by a BT speaker. It's not easy to do.

Yes, that is my intension. 
What do you think, if we provide the media with the required samples. Would this make things much easier?

closed 2021-03-05 by stale[bot]