schreibfaul1/ESP32-audioI2S #86: Reason for UTF-8 to ASCII Filepath conversion?

grch87, 2020-12-20:
Hi,
I'm just wondering, why there is an UTF-8 to ASCII conversion for the filepath in the connecttoFS function?

My problem is that I want to play german tracks, which contain not ASCII conform characters (e.g. Ä,Ö,Ü,ß). When I call connecttoFS with such a track, the filepath got converted and the filesystem is no longer able to find the file. When I comment out this conversion section everything is working fine.
It seems that is more like a renaming.

I don't fully understand the conversion, as, for instance, an 'Ä' character is in UTF-8 represented in decimal as 228 and after the conversion it is 132, so '"'.

Is this conversion really necessary, as the calling application of connecttoFS is pasing apart from the filepath as well the reference of the filesystem? Thus it should be in the responsibility of the application developer to pass the correct format of the filepath.
If it is necessary, would it be possible to optionally deactivate the conversion via an compiler swich?

Apart from that issue, I'm quite happy with this nice library and I want to say thank you :)

schreibfaul1, 2020-12-22:
You are right, the conversion UTF8 to ASCII does not work properly. Basically the file name only consists of ASCII characters. UTF8 (C2xx or C3xx) is not recognized. Now files with names like "Straße.wav" or "Äpfel.mp3" played without errors.

grch87, 2020-12-23:
Great, thank you for looking into it!

CelliesProjects, 2020-12-23:
So I wrote al that code for nothing?
```
String htmlEntities(const char* plaintext) {
  String result{};
  uint32_t cnt{0};
  while (plaintext[cnt] != 0) {
    if (plaintext[cnt] > 0x7F || plaintext[cnt] < 0x20) {
      switch (plaintext[cnt]) {

        case 0xC2 : {
            const uint8_t firstByte = plaintext[cnt];
            cnt++;
            const uint8_t secondByte = plaintext[cnt];
            switch (secondByte) {
              case 0xA0 ... 0xBF : {
                  result.concat((char)firstByte);
                  result.concat((char)secondByte);
                }
                break;
              default: {
                result.concat("?");
                ESP_LOGE(TAG, "Invalid 16-bit utf8 sequence. Dropped 2 bytes.");
              }
            }
          }
          break;

        case 0xC3 : {
            const uint8_t firstByte = plaintext[cnt];
            cnt++;
            const uint8_t secondByte = plaintext[cnt];
            switch (secondByte) {
              case 0x80 ... 0xBF : {
                  result.concat((char)firstByte);
                  result.concat((char)secondByte);
                }
                break;
              default: {
                result.concat("?");
                ESP_LOGE(TAG, "Invalid 16-bit utf8 sequence. Dropped 2 bytes.");
              }
            }
          }
          break;

        case 0xC9 :
          result.concat("&Eacute;"); // É
          break;

        case 0xE1 :
          result.concat("&aacute;"); // á
          break;

        case 0xE4 :
          result.concat("&auml;"); // ä
          break;

        case 0xE7 :
          result.concat("&ccedil;"); // ç
          break;

        case 0xE8 :
          result.concat("&egrave;"); // è
          break;

        case 0xE9 :
          result.concat("&eacute;"); // é
          break;

        case 0xEA :
          result.concat("&ecirc;"); // ê
          break;

        case 0xEB :
          result.concat("&euml;"); // ë
          break;

        case 0xED :
          result.concat("&iacute;"); // í
          break;

        case 0xF3 :
          result.concat("&oacute; "); // ó
          break;

        case 0xF6 :
          result.concat("&ouml;"); // ö
          break;

        case 0xFC :
          result.concat("&uuml;"); // ü
          break;

        default : result.concat("?");
          ESP_LOGE(TAG, "ERROR: Unhandled char 0x%x", plaintext[cnt]);
      }
    }
    else
      result.concat(plaintext[cnt]);
    cnt++;
  }
  ESP_LOGD(TAG, "Input str: %s", plaintext);
  ESP_LOGD(TAG, "Returning html encoded str: %s", result.c_str());
  return result;
}
```
Just joking.

schreibfaul1, 2020-12-23:
Hi Cellie,
there was still old code from my VS1053_ext lib used (for converting ISO8895 and not UTF-8) and after two years the user grch87 found it !! :-)

grch87, 2021-01-27:
Hi,
would it be possible to have the option via a compiler switch to deactivate the utf-8 conversion completely?
The reason is that I want to play as well files, which have an UTF-8 encoded filename. The conversion is more or less changing the filename and the Filesystem is no longer able to find the file. When I remove the conversion it is working.
I can prepare a pull request if you like.

schreibfaul1, 2021-01-27:
Can it be that ISO-8859 is set as "text file encoding" instead of UTF-8 in the IDE?

grch87, 2021-01-28:
The text file encoding in the IDE shouldn't influence the filename when I copy files onto the SD card, or did I miss something?
I contribute to the ESP32 Tonuino Project and there are different possibilities how to copy audio files onto the SD card:
- Inserting the SD into the PC
- via FTP
- via HTTP
The ESP32 Tonuino users have as well different OS Win/Mac/Linux.
So in my opinion the filename encoding depends on the "how are files copied onto the SD card". For FTP and HTTP the encoding can be influenced in a way, but not when the user is directly copying the files.

If there is an audio file with an UTF-8 encoded filename it can't be played via the this audio lib, because of this conversion. Without the conversion it is working.
The ESP32 Tonuino is getting the filename of an audio file from the filesystem. It is the same filesystem in the connecttofs function which is opening the file. A conversion filename before opening the file causes problems, because the audio file can't be found.

schreibfaul1, 2021-01-28:
File names are saved as UTF-8. In file names like 'ö.mp3' the 'ö' is interpreted as 0xC3 0xB6. Unfortunately the Arduino Framework doesn't understand that.
The 'ö' here is the ASCII value 0x94 (148). Therefore the conversion is necessary. If the IDE is set to ASCII, the conversion is not required.
Or the Tonuino SW already does something with the file name?

grch87, 2021-01-28:
Thanks for your patience. I guess we missed so far each others point.
But maybe I got now yours :)
Is it possible that the conversion is used, that the filename is correctly printed on the console? If that is the case there might be a bug.
Here a pseudo code of the current conntecttofs function:

bool Audio::connecttoFS(fs::FS &fs, const char* file) {
  String s_file = file;
  path= convertToASCII(s_file);
  print("Reading file %s", s_file);
  audiofile  = fs.open(path);
  ...
}

I guess "path" shall be used for printing the filename and s_file for opening the file? So the other way arround?

schreibfaul1, 2021-01-28:
The file for the test is called 'März.wav'.

    log_i("file=%s", file);
    log_i("path=%s", path);
    log_i("m_audioName=%s", m_audioName);

And that's the result in the serial monitor (Eclipse IDE with Sloeber plugin):

[image]

Which filenames do not work? Are they sure names with special characters?

grch87, 2021-01-28:
the current implementation uses path to open the file: fs.open(path)
with an UTF-8 encoded audio file März.mp3, I get the following:
[image]

when I do: fs.open(file) it is working
[image]

schreibfaul1, 2021-01-28:
when I do: fs.open(file); I get an error:

[image]

You can use UTF-8 filenames, I only use ASCII. Why is that? Do you use a special SD library?

grch87, 2021-01-28:
I use the standard "FS.h" and "SD_MMC.h" libs. 
Via FTP, for instance, it is possible to load files onto the SD card with UTF-8 filename encoding. The good thing is, that I get the filename of an audio file I want to play from the filesystem. 
Something like that:
`String filename = file.name();`
and call then connecttofs. So basically I don't require this conversion, as I call connecttofs already with the correct filename.

So would it be possible to deactivate this conversion optionally? :)

schreibfaul1, 2021-01-28:
Hi grch87,
then the problem is:
SD.h uses ASCII
SD_MMC.h uses UTF-8

and the solution would be:
If (SD_MMC) fs.open (file);
Else fs.open (path);

Unfortunately I still have no idea how I can distinguish SD from SD_MMC

grch87, 2021-01-29:
Hi, thank you for the commit.
If I may speak honestly, it looks quite complicated and I'm not sure if it creates more issues than it will solve.

Wouldn't be something like that simpler?

if(fs.exists(path)) {
  audiofile = fs.open(path);
} else {
  audiofile = fs.open(s_file);
}

If the filesystem can't find a file "path", the fallback is "s_file".
This should even cover more cases and you don't have to take care about the storage interface or the defined codepage. 
Just want to make your life easier!

schreibfaul1, 2021-01-30:
Thank you very much, your solution is much better :-))

grch87, 2021-01-30:
Wow, thank you for the quick response and implementation :)

closed 2021-01-30 by grch87