schreibfaul1/ESP32-audioI2S #87: TTGO TM ESP32

A-Cheikh, 2020-12-21:
Please, I am using TTGO TM board : https://github.com/LilyGO/TTGO-TM-ESP32, and I am trying the last code above of audio and that is not work , I dont know if I make mistake in I2S pins
#define I2S_DOUT 19 //DIN
#define I2S_BCLK 26 //BCK
#define I2S_LRC 25 //LCK
[image]

CelliesProjects, 2020-12-22:
Your pins should be good like that.

I have that board working with the example. But I never play from SD so can not help you there.
Maybe you forgot the `audio.loop()` in the main loop?

```
#include "Arduino.h"
#include "WiFiMulti.h"
#include "Audio.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define I2S_DOUT      19
#define I2S_BCLK      26
#define I2S_LRC       25

Audio audio;
WiFiMulti wifiMulti;

String ssid =     "xxx";
String password = "xxx";

void setup() {
    WiFi.mode(WIFI_STA);
    wifiMulti.addAP(ssid.c_str(), password.c_str());
    wifiMulti.run();
    if(WiFi.status() != WL_CONNECTED){
        WiFi.disconnect(true);
        wifiMulti.run();
      }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(12); // 0...21

    //audio.connecttoFS(SD, "/320k_test.mp3");
    //audio.connecttohost("www.wdr.de/wdrlive/media/einslive.m3u");
    //audio.connecttohost("dg-ais-eco-http-fra-eco-cdn.cast.addradio.de/hellwegradio/west/mp3/high");
    audio.connecttohost("http://macslons-irish-pub-radio.com/media.asx");
    //audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");
}

void loop()
{
    audio.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}
```

marc-gist, 2020-12-23:
Post more of your code; I have similar board running with SD playing fine.

A-Cheikh, 2020-12-26:
Finnaly it works thanks you all of you guys, but there is another problems , here is the program:
#include "Arduino.h"
//#include "WiFi.h"
#include "Audio.h"
#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS         13
#define SPI_MOSI      15
#define SPI_MISO       2
#define SPI_SCK       14
#define I2S_DOUT      19  //DIN
#define I2S_BCLK      26  //BCK
#define I2S_LRC       25  //LCK

Audio audio;

//String ssid =     "*******";
//String password = "*******";

void setup() {
    pinMode(SD_CS, OUTPUT);     
    digitalWrite(SD_CS, HIGH);
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
    Serial.begin(115200);
    Serial.println("Debut test carte");
    if (!SD.begin(SD_CS)) {
      Serial.println("Card Mount Failed");
    }
    Serial.println("Fin test carte");
    //WiFi.disconnect();
    //WiFi.mode(WIFI_STA);
    //WiFi.begin(ssid.c_str(), password.c_str());
    //while (WiFi.status() != WL_CONNECTED) delay(1500);
    //audio.connecttoFS(SD, "/001.mp3");
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.setVolume(21); // 0...21
    Serial.println("Debut test son");
    audio.connecttoFS(SD, "/001.mp3");
    Serial.println("Fin test son");
}

void loop()
{
    audio.loop();
}

 but while reading images from SD CARD it takes 56% from programm memory and that's so big, I want to add another functionalities of audio .. but it is insufficient , is there any method for that to not waste a lot of prog memory?

CelliesProjects, 2020-12-26:
Try setting the maxfiles to 1 in `SD.begin()`.

Something like 

`SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI, 4000000, "/sd", 1);`

Check https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/SD.h#L31

Now there is only 1 filebuffer. How much you can reclaim I don't know from the top of my head.

A-Cheikh, 2020-12-26:
Hi, thanks for your comments , but it could be something wrong here, it takes always 56%
C:\Users\achraf\Desktop\ard.png

schreibfaul1, 2020-12-26:
The library needs the storage space. You can use 'Huge App' as a partition table instead of 'Default'. The available program memory then increases from 1310720 to 3145728 bytes.

@Cellie
````
SD.begin(SD_CS, SPI, 4000000, "/sd", 1);
````
I didn't know it at all, that saves about 10kB heap for me, thanks!

A-Cheikh, 2020-12-26:
Thanks @schreibfaul1 for your comment, you say **_"You can use 'Huge App' as a partition table instead of 'Default'. The available program memory then increases from 1310720 to 3145728 bytes".**_ , please How can I do it?

schreibfaul1, 2020-12-26:
Arduino IDE:
Tools / Partitions Scheme / Huge App (3MB no OTA, 1MB SPIFFS)

A-Cheikh, 2020-12-26:
Thank you very much @schreibfaul1 @CelliesProjects  it works soo good, now the prog memory is minimized to 26%

CelliesProjects, 2020-12-27:
I should learn to read. Progmem != dynamic mem.

closed 2021-03-12 by stale[bot]