schreibfaul1/ESP32-audioI2S #88: playing wave files from SD card cause core dump

marc-gist, 2020-12-22:
```
Playing File...
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x800d239e  A1      : 0x3ffb1c90  
A2      : 0x3ffc1210  A3      : 0x00000010  A4      : 0x3ffc1340  A5      : 0x3ffc3510  
A6      : 0x00000001  A7      : 0x3ffc1288  A8      : 0x800dc07a  A9      : 0x0000ac44  
A10     : 0x3ffc1340  A11     : 0x0000ac44  A12     : 0x000ac440  A13     : 0x00000001  
A14     : 0x00000001  A15     : 0x00000010  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb1c90 0x400d239b:0x3ffb1d70 0x400d3ba9:0x3ffb1da0 0x4018598b:0x3ffb1f30 0x400d4ffa:0x3ffb1f50 0x400d47e2:0x3ffb1f90 0x400f2ae1:0x3ffb1fb0 0x40089785:0x3ffb1fd0

Rebooting...
```

using:
`audio.connecttoFS(SD, file);`

where file is a String converted from MQTT: ex:
`/galaxy_invaders/player_shoot.wav`

schreibfaul1, 2020-12-22:
Is there a link so I can test 'player_shoot.wav'?

marc-gist, 2020-12-22:
[player_shoot.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/5732433/player_shoot.zip)

schreibfaul1, 2020-12-23:
The file is played with me without any problems
````
audio_info: buffers freed, free Heap: 144220 bytes
audio_info: Reading file: "player_shoot.wav"
audio_info: FormatCode=1
audio_info: Channel=1
audio_info: SampleRate=44100
audio_info: DataRate=88200
audio_info: DataBlockSize=2
audio_info: BitsPerSample=16
audio_info: BitRate=705600
audio_info: DataLength=35334
audio_info: stream ready
audio_info: End of file "player_shoot.wav"
````
Is it possible that too much time is used in the loop? InstrFetchProhibited indicates this.

marc-gist, 2020-12-23:
@schreibfaul1 possible i guess. i'm using someone elses code that also drives an LCD at the same time, so perhaps it is.
However, interesting that MP3 plays fine, and 6 wave files wont.  but if i convert them to MP3 they do fine... perhaps due to the compression its less data, therefore less time in the loop to process? i would have though it needs to decompress... 

I'll see if I can clean up the code and make it faster to loop.

marc-gist, 2020-12-23:
@schreibfaul1 good call.  that's exactly what it was :( 
Screen was getting updated every 500ms, changed it to 800ms and WAV files  play fine now (or at least the really short ones I have).

closed 2020-12-23 by marc-gist