schreibfaul1/ESP32-audioI2S #91: Feature request: Microphone Support

NonaSuomy, 2020-12-29:
Are you able to add support for the microphone feature of these chipsets?

Thank you.

CelliesProjects, 2021-02-12:
The UDA1334A dac does not have an mic input.
If you have a WM8978 dac it is already supported in the [WM8978 driver](https://github.com/CelliesProjects/wm8978-esp32).

Not sure about other dacs.

NonaSuomy, 2021-03-01:
Is there any examples you know of for the WM8978?

CelliesProjects, 2021-03-02:
You can use the [Espressif examples](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#application-example) as a base.

Set up the wm8978 and use [i2s_read](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv48i2s_read10i2s_port_tPv6size_tP6size_t10TickType_t) to get some data.
Most ESP-IDF calls will run happily in Arduino land.

closed 2021-06-09 by stale[bot]