schreibfaul1/ESP32-audioI2S #96: Change in SetFilePos

WolfgangFfm, 2021-01-09:
Hi, using SetFilePos for start playing another part of a song makes the "getAutioCurrentTime" go wrong.
A little change in the function setFilePos makes it work:
I added for safety: 
    if (pos<m_id3Size) pos=m_id3Size;

I addes for the right CurrentTime:
    if (m_avr_bitrate) m_audioCurrentTime=(pos-m_id3Size) * 8 / m_avr_bitrate;

Now moving inside a song works!

For setting the new position in the song I used:

  audio.setFilePos(audio.getFileSize()*NewPercentage/100);

schreibfaul1, 2021-01-09:
Thanks for the helpful hints. I hope the audioCurrentTime is now calculated correctly.

WolfgangFfm, 2021-01-26:
Hi Wolle,

ich habe gesehen, dass du in die audio-Lib eine Filtermöglichkeit  
eingebaut hast. Warum? Die Filtermöglichkeit ist in dem AC101 bereits  
implementiert. es fehlt in der AC101-Lib (https://github.com/Yveaux/AC101)  
nur die Ansteuerung. Ich werden in den nächsten Tagen mal anfangen die  
AC101-Lib zunächst für mich zu erweitern.

Ich habe hier für meine Versuche die AT-Thinker Audio-Kit Platine mit dem  
ESP32-A1S der den AC101 mit drauf hat. Diese verwende ich mit meinem  
Web-Interface als MP3 / Radio Player. Damit die MP3 Lieder nicht immer in  
der gleichen Rehenfolge abgespielt werden, habe ich ein "zufälliges  
Überspringen" von Direinträgen in die FS-Lib integriet. Beim Erweitern der  
Lib habe ich auch eine Suchfunktion und ein "Zählen der Lieder" auf der  
MicroSd implementiert. Wenn du an den "paar" Zeilen Code interessiert  
bist, dann schicke ich sie dir mal.

Die von mir in deine Lib reingebrachten Zeilen funktionieren sehr gut. Die  
Berechnung stimmt zwar nicht exakt, aber der Unterschied ist nicht zu  
merken.  (bei der Umrechnung LiedZeit->FilePos wird der Header nicht  
vorher von der FileSize abgezogen)

schreibfaul1, 2021-01-26:
Hallo Wolfgang,

gut zu wissen, dass man mit dem A1S auch eine Klangregelung realisieren kann. Ich hatte im Mikrocontrollerforum ein Hinweis auf die AC101-Lib gefunden, diese eingebaut und die lief auf Anhieb. Daher habe ich mich nicht weiter mit dem AC101 beschäftigt. Es wäre toll, wenn Du die Lib erweitern könntest. Mit den IIR Filtern kenne ich mich nicht wirklich gut aus
das war mehr ein erster Versuch, zumal es nicht viele praktische Beispiele zum Thema gibt, für den ESP32 habe ich gar nichts gefunden. Und das digitale Filer kann man noch für die "normalen" externen DACs verwenden.
Ich bin an jeder Fehlerbereinigung und Verbesserung interessiert und für alle Vorschläge dankbar.
Das mit der Umrechnung LiedZeit->FilePos gucke ich mir in den nächsten Tagen an.

bis dann, 
vG Wolle

WolfgangFfm, 2021-01-27:
Hi Wolle,

mit meinem "Die Filtermöglichkeit ist in dem AC101 bereits implementiert"  
habe ich mich etwas weit aus dem Fenster gelehnt: es gibt dort nur eine  
LPF (Low pass Filter) mit 2 Coeffizienten. Ich werde mit denen gleich mal  
herumexperimentieren, aber eine Klangregelung ist damit sicher nicht  
möglich. Also solltest du dafür doch deinen Ansatz weiter verfolgen.

Ich hatte mich nur in den AC101 reingekniet, weil ich den DRC (Dynamic  
Range Control) zum Laufen bringen möchte. Manche meiner Lieder sind laut  
und manche leise, deswegen wäre eine "Lautstärkeautomatik" wünschenswert.  
Aber ich fürchte es ist keine AGC (Auto Gain Control) damit zu  
realisieren, sondern eher nur eine Dynamikkompression.

Ich melde mich wieder. Bis dann bleib gesund.

closed 2021-06-01 by stale[bot]