schreibfaul1/ESP32-audioI2S #99: Constructor and setPinout

MarkusLange, 2021-01-26:
Why did I need to set the setPinout in the setup() part when the Construktor Audio allready did it? 

`Audio(const uint8_t BCLK=26, const uint8_t LRC=25, const uint8_t DOUT=27);`

And why is the Lib not working when I remove 

`audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);`

from the setup()?

schreibfaul1, 2021-01-26:
Hi Markus,
almost any GPIO can be used for I2S. With the ESP32-A1S the pins are internally wired and the default must be used. I had assigned the pins differently than in the example. I corrected that. Calling the **Audio: audio;** object is now possible without **setPinout (...)**.
[image]

MarkusLange, 2021-02-08:
Thanks for the explanation, that makes some parts clearer, but for me I run in an similar trouble with my Board (Lolin D32 Pro) I used Pins:
```
#define I2S_BCLK       2
#define I2S_LRC       15
#define I2S_DOUT       0
```
What was not working with the Audio audio; objekt by setting the Pins, so I dig deeper into your files:

In
`Audio(const uint8_t BCLK=26, const uint8_t LRC=25, const uint8_t DOUT=27);`
3 Pins are set all predefined for the ESP32-A1S and in
`bool setPinout(uint8_t BCLK, uint8_t LRC, uint8_t DOUT, int8_t DIN=I2S_PIN_NO_CHANGE);`
4 Pins are set with DIN predefined as "-1"
Now the the interesting part, if I set my Pins in the Construktor it was not working, with setPinout it does, so I see this:
```
Audio::Audio(const uint8_t BCLK, const uint8_t LRC, const uint8_t DOUT) {
...
    m_BCLK=BCLK;                       // Bit Clock
    m_LRC=LRC;                         // Left/Right Clock
    m_DOUT=DOUT;                       // Data Out
    setPinout(m_BCLK, m_LRC, m_DOUT, m_DIN);
```
the Construktor gives 3 Pins and setPinout gets 4 Pins on the other hand also setPinout does not care, m_DIN must be set anyway in Audio.h all set
```
...
    int8_t          m_balance = 0;                  // -16 (mute left) ... +16 (mute right)
    int8_t          m_DIN=0;                        // Data In, can be negative if unused (I2S_PIN_NO_CHANGE is -1)
    uint8_t         m_rev=0;                        // revision, ID3 version
    uint8_t         m_BCLK=0;                       // Bit Clock
    uint8_t         m_LRC=0;                        // Left/Right Clock
    uint8_t         m_DOUT=0;                       // Data Out
    uint8_t         m_vol=64;                       // volume
...
```
the Header deliver the missing values. In short if I set the Pins with Audio::Audio m_DIN is set to "0" allways can't change. It will collide if I use Pin "0" somewhere. So I have made some changes to this:
**Construktor**:
clean predefines and add DIN (in Audio.h and Audio.cpp)
```
Audio(const uint8_t BCLK, const uint8_t LRC, const uint8_t DOUT, const int8_t DIN = I2S_PIN_NO_CHANGE);
...
    m_BCLK = BCLK;                       // Bit Clock
    m_LRC  = LRC;                        // Left/Right Clock
    m_DOUT = DOUT;                       // Data Out
   m_DIN  = DIN;                        // Data In, can be negative if unused (I2S_PIN_NO_CHANGE is -1)
    setPinout(m_BCLK, m_LRC, m_DOUT, m_DIN);
```
**Header File**:
set all Pins variables without value
```
    int8_t          m_balance             = 0;          // -16 (mute left) ... +16 (mute right)
    int8_t          m_DIN;                              // Data In, can be negative if unused (I2S_PIN_NO_CHANGE is -1)
    uint8_t         m_rev                 = 0;          // revision, ID3 version
    uint8_t         m_BCLK;                             // Bit Clock
    uint8_t         m_LRC;                              // Left/Right Clock
    uint8_t         m_DOUT;                             // Data Out
    uint8_t         m_vol                 = 64;         // volume
```
Now I have to set at least 3 Pins inside the Construktor to get proper working code and get DIN set to "-1" if not set so it does not collide with the use of PIN "0"

CelliesProjects, 2021-02-09:
Do `m_BCLK`, `m_LRC`, `m_DOUT`, and `m_DIN` really have to be saved globally?

They are passed in and used in `setPinout` and in the ctor and once that has happened they are not really needed any more.

schreibfaul1, 2021-02-10:
Hello Markus, hello Cellie,
quite right, m_DIN = 0 is unnecessary and blocks the GPIO0. And m_DOUT, m_BLCK .. are no longer global variables, but are now embedded in a structure.

CelliesProjects, 2021-02-13:
I did some testing on a m5stack with a node base and still think the code has some issues.

The empty ctor `Audio audio` (as in the example) still uses the preset values and messes up in basically the same way as discussed above.
When an empty audio instance is created, it sets (messes up) the pins.

Using `setPinout` again does unfortunately not unmess the set pins.

And it is confusing. You set an empty ctor but get the default values.

My proposal would be to get rid of the empty ctor altogether. (So no default values)
But this would break the example and probably other existing code.

schreibfaul1, 2021-02-13:
Hi Cellie,
only now do I know what the problem is. I pre-assigned the outputs for the A1S in the constructor. DOUT has the value 25. Now I have set DOUT to pin 14 in setPinout. That works so far. Unfortunately DOUT is then on pin 14 and pin 25. The old assignment is not revoked. I don't see any other way to get pin 25 free except to delete the pre-assignment in the constructor

MarkusLange, 2021-02-13:
That was the way I had to choose to get my problem solved, if you clean the constructor.

CelliesProjects, 2021-02-13:
If you have a constructor where all values must be entered then also maybe 'setPinout' does not need to be public.

CelliesProjects, 2021-02-16:
Thanks. Fixed the issue for me.

closed 2021-06-01 by stale[bot]