schreibfaul1/ESP32-audioI2S #103: Audio stutter in continuos loop

Theexternaldisk, 2021-02-02:
When playing an audio file in a continuous loop it will stutter every now and again. 
On the scope it looks like the data is starting and stopping causing the audio to stutter.
Any ideas what could cause this?
I'm playing a short audio file (5 sec) from an SD card using the (modified to loop) standard example code.
Longer files also seem to have this problem and it seems to happen after a few loops right around or after the loop point.
The data Signal:
[image]
One part of the (intermittent) data signal:
[image]

schreibfaul1, 2021-02-03:
Hi Theexternaldisk,
the InBuff was previously filled with the beginning of the file at the end of an audio file, as long as there was still space. The mp3 decoder then did not find the syncword for the next beginning. Now the audio file is played completely, the InBuff is reset and the file is read in again. That seems to work better.

Theexternaldisk, 2021-02-03:
This did not fix the stutter unfortunately.
I've done some more tests.
The stutter interval seems to occur at (semi) fixed intervals. With one audio file it occurs around the loop point of the 10th to 11th loop.
On another shorter file it occurs around the loop point of the 39th to 40th loop.
And it also seems to happen when not using the loop function. So looping by loading the file again after it finished playing.
I'll have to look into the code and see why/where it goes wrong.
I can share my audio files if you are interested.

Theexternaldisk, 2021-02-04:
The stutter seems to occur because reading from the SD-card takes substantially longer than usual.
Usually it takes between 0-14msec and during the stutter it is between 29-147msec with jumps to 0msec.
To find this I measured the time of line 1159 which reads from the SD card.
So it does not seem to be a problem in the library after all.
Any closing thoughts?
I should try another SD-card of course but any additional tips are welcomed.

schreibfaul1, 2021-02-04:
The SD card won't be the problem. Maybe the voltage differs from 3.3V. If there are still GPIOs free, the SD card can be connected as SD_MMC.

biologist79, 2021-02-04:
@Theexternaldisk When it comes to SD, make sure to solder all the stuff. When I started by project, even with short jumperwires, I had issues with ugly sound from SD wheras via webstream it was fine. After soldering all the stuff it was gone completely.

Theexternaldisk, 2021-02-05:
Thanks for the advice people.
I've had it running on another SD card for a while now and the stutter seems to be less frequent.
During the stutter the SD card is sending back no data. But I'll have to investigate this further, still trying to find my way around this oscilloscoop.
Or perhaps there is a way to see why the SD card read command takes longer than usual during the stutter?
@schreibfaul1 the vcc is at 3.3v (don't know what the maximum ripple voltage should be)
I still have to try the SD_mcc
@biologist79 it is a custom pcb which has everything surface mount and soldered properly.
The sound quality over all is fine I just have the occasional stutter

closed 2021-06-01 by stale[bot]