schreibfaul1/ESP32-audioI2S #104: Compile Esp32 ?

alwassmer, 2021-02-04:
I can not get this to compile for esp32.
I get this error 

C:\Users\Al\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In constructor 'Audio::Audio(uint8_t, uint8_t, uint8_t)':

C:\Users\Al\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:134:17: error: 'class WiFiClientSecure' has no member named 'setInsecure'

    clientsecure.setInsecure();  // if that can't be resolved update to ESP32 Arduino version 1.0.5-rc05 or higher

can you point me in the right direction ?

audio.h  seems to be my problem
                                                                              Al

schreibfaul1, 2021-02-04:
You need a newer ESP32-Arduino version. The update can be carried out in the Boads Manager in the Arduino IDE.
[image]

biologist79, 2021-02-04:
As I've just learned yesterday, in Platformio this can be done this way via platformio.ini:
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#1.0.5-rc6

closed 2021-02-10 by alwassmer