schreibfaul1/ESP32-audioI2S #106: Play from SD card not working anymore.

christi-ko, 2021-02-05:
Just updated to newest Commit. Seems like it's not playing from SD card anymore. Log looks like this:

`DEBUG:        buffers freed, free Heap: 189608 bytes
DEBUG:        Reading file: "The Weeknd - Blinding Lights (Official Music Video).mp3"
DEBUG:        MP3Decoder has been initialized, free Heap: 165860 bytes
DEBUG:        stream ready
DEBUG:        ID3 framesSize=15108
DEBUG:        ID3 version=4
DEBUG:        ID3 normal frames
DEBUG:        syncword found at pos 183
DEBUG:        syncword found at pos 0
DEBUG:        End of file "The Weeknd - Blinding Lights (Official Music Video).mp3"
audio_eof:   The Weeknd - Blinding Lights (Official Music Video).mp3Button Short`

The EOF is shown about 5s after start.

schreibfaul1, 2021-02-05:
Hi christi-ko,
it is not determined by the file size when the end is reached, but EOF is called as soon as nothing is read from the file. Then the variable bytesAddedToBuffer == 0 in the sub processLocalFile (). An error could have occurred when reading out the ID3 metadata. My test files are working correctly. Is there a link to the audio file for testing?

christi-ko, 2021-02-05:
OK, i will test first with your test files and come back with an answer. However, i tested the previous commit (i think 29.12.2020, prior to the Arduino core 1.0.5rc5 required) and it worked like a charm. The only thing i added also is that i use mono option also. But i will test that too ...
regards
christi-ko

christi-ko, 2021-02-06:
Hi Wolle,
just did the testing as described above with following results:
1.) Your Test file is working
2.) MONO or Stereo has no impact on working or not (as expected).
3.) Preliminary result: Its linked to my mp3 file (not only one of them). These files worked with previous versions of this library. 
==> Now i don't know what to do. I attached one example, maybe you could point me to a solution.

regards
Christian

schreibfaul1, 2021-02-06:
Hi Christian,
In earlier versions of the Lib, only the ID3 header size was determined by a server for mp3 files and the header was skipped. The information in the header is now read out (as with local files). So far this has worked fine with my test files. The attached file 'Helicopter 117' differs in the ID3V2 version. The ID3V2.4 calculates the ID3 frame size differently than ID3V2.3. Now it will work.

regards
Wolle

christi-ko, 2021-02-10:
Hi Wolle, 
thanks for your great support. I tested it with my files and everything works as expected. Really great library which you created. I'm working on a Webradio with SD Card which is using your TFT Library in combination with a rotary encoder to browse the menu. For sure not that good like your Webradio. If your interested in using it as an example code i could forward it to you once it's in a more stable state, please let me know ....
regards
Christian

christi-ko, 2021-02-13:
Issue is closed and verified.

closed 2021-02-13 by christi-ko