schreibfaul1/ESP32-audioI2S #109: Stream Radio Station

alwassmer, 2021-02-10:
Hi
I have this  URL for  KMOX Radio St Louis .It works in VLC player under open network streams.  This is the URL        http://23103.live.streamtheworld.com/KMOXAMAAC
Can not get it to work in esp32-audioI2S it plays nothing.
Should this work. or is it not compatible ?

Radio.com hoist KMOX . Is it  possible to play Radio.Com Streams?

                                                                   Thanks Al Wassmer

alwassmer, 2021-02-11:
Some more info!  this stream is a mp4a format. here is a link on it if you want to add it to your library.               https://www.filetypeadvisor.com/extension/mp4a    

                                                                                                                       Al

schreibfaul1, 2021-02-11:
Hi alwassmer,
M4a cannot be played. This is not a pure AAC, the audio data is packed in containers. And the header must be read out in order to determine the beginning of the audio data.

alwassmer, 2021-02-12:
Thanks for the reply!
Do you know if the VS1053 decoder will play this? and if so do you have a project that supports the vs1053.

schreibfaul1, 2021-02-13:
Hi alwassmer,
m4a files are too complex for this library. It is difficult to find the beginning of the audio container and the AAC data are in RAW format or are difficult to process by the AAC decoder. I was able to play high quality m4a files with the VS1053 using my lib ESP32-vs1053_ext.

alwassmer, 2021-02-13:
Ok ! Thanks for the info

closed 2021-06-01 by stale[bot]