schreibfaul1/ESP32-audioI2S #110: How to play and store audio file from Google translate url?

playground, 2021-02-11:
Hi, I'm trying to play the audio on Atom Echo Smart Speaker Development Kit, for example: play the audio return from https://translate.google.com/translate_tts?ie=UTF-8&q=Hello%2C+World!&tl=en&client=tw-ob&ttsspeed=1 and and be able to store it for playback later?

schreibfaul1, 2021-02-11:
Hi, you can run
```connecttospeech("Hello World!", "en");```
and listen to the mp3 stream, recording does not work, the lib is not a sound recorder

closed 2021-02-11 by schreibfaul1

playground, 2021-02-12:
Thanks @schreibfaul1, I'm getting this error 

platformio.ini
`[env:m5stack-core-esp32]
platform = espressif32
board = m5stack-core-esp32
framework = arduino
upload_speed = 115200
monitor_speed = 115200
lib_deps = 
	hieromon/AutoConnect@^1.2.2
	me-no-dev/ESP Async WebServer@^1.2.3
	celliesprojects/wm8978-esp32@^1.0.2`

`In file included from src/main.cpp:5:0:
.pio/libdeps/m5stack-core-esp32/Audio/Audio.h:31:2: error: #error "Teensyduino version 1.20 or later is required to compile the Audio library."
 #error "Teensyduino version 1.20 or later is required to compile the Audio library."
  ^
.pio/libdeps/m5stack-core-esp32/Audio/Audio.h:37:24: fatal error: DMAChannel.h: No such file or directory

********************************************************************
* Looking for DMAChannel.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:DMAChannel.h"
* Web  > https://platformio.org/lib/search?query=header:DMAChannel.h
*
********************************************************************

compilation terminated.
Compiling .pio/build/m5stack-core-esp32/lib3bc/Audio/analyze_fft1024.cpp.o
Compiling .pio/build/m5stack-core-esp32/lib3bc/Audio/analyze_fft256.cpp.o
Compiling .pio/build/m5stack-core-esp32/lib3bc/Audio/analyze_peak.cpp.o
Compiling .pio/build/m5stack-core-esp32/lib3bc/Audio/analyze_print.cpp.o
Compiling .pio/build/m5stack-core-esp32/lib3bc/Audio/analyze_tonedetect.cpp.o
In file included from .pio/libdeps/m5stack-core-esp32/Audio/analyze_fft1024.cpp:27:0:
.pio/libdeps/m5stack-core-esp32/Audio/analyze_fft1024.h:30:25: fatal error: AudioStream.h: No such file or directory
`

playground, 2021-02-12:
Thanks @schreibfaul1, I'm using wm8978-esp32 and it's throwing this error https://github.com/CelliesProjects/wm8978-esp32/issues/6