schreibfaul1/ESP32-audioI2S #111: Integrate new SdFat Library instead of SD

christi-ko, 2021-02-11:
Based on several discussion which can be found on the internet, the SD Library is kind of outdated and will take very long if you want to list files on the SD card. I am currently facing exactly this issue and listing about 100 files takes several second (>10s). The proposed solution which can be found is to change for the newer SdFat Library, which is the forward developed version of the SD library. However there are changes in the library, which are not compatible with this library. There are several initialization of datatype used which are directly linked to the SD library. Would it be possible to have that as an option in order to reduce SD loading times. 

regards
Christian

schreibfaul1, 2021-02-11:
Hi Christian,
the call audio.connecttoFS(FFat, 'test.wav'); will work now. I don't think this is the right way, the memory size is limited (1.5 ... 2 MB). It is better to use the SD_MMC lib instead of SD. The 1 wire connection is good, 4 wire brings about 10% more speed. And 4 wire does not run without changes to WROOM and WROVER. What to do:
MISO becomes D0 on GPIO2
MOSI becomes CMD on GPIO15
SCLK becomes CLK on GPIO14
You can only use the GPIOs provided for this purpose, and switch PullUp to GPIO2.
PinMode (2, INPUT_PULLUP);

regards
Wolle

christi-ko, 2021-02-11:
Hi Wolle,
thanks for your Feedback. However i am not sure if we did talk about the same thing. My request was about the following library:
https://github.com/greiman/SdFat
this should not use any other interface nor more memory, or did i get that wrong?

However i looked into your suggestion about using the SD_MMC lib instead. Do i need to connect something in a different way to make it work? I tried it to change in code like this. 

From:
`#include "SD.h"
...
 SD.begin(SD_CS);
`
To:
`#include "SD_MMC.h"
...
SD_MMC.begin();
`
However i did get the failure that no SD card is mounted. Do i need to wire things up in a different way? Can you help me here? Maybe it would be wort to add a 2nd example in your library showing how SD_MMC is used.

regards
Christian

schreibfaul1, 2021-02-12:
I misinterpreted that, SdFat is not FFat. SDFat will not be easy to integrate, a lot of manual work is required.
I'm not sure if SD_MMC can list a lot of files in a short amount of time, but faster than the SPI connection. Card adapters without electronics are suitable, just solder 5 wires (GND, 3.3V, D0, CMD, CLK). I use this adapter:
[image]

christi-ko, 2021-02-12:
I tried it the whole morning, but i can't get the SD_MMC in one line configuration to work. First i had issues flashing due to GPIO02 and after disconnection it i was able to flash it. However still didn't work. 

`E (45628) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
[E][SD_MMC.cpp:85] begin(): Failed to initialize the card (265). Make sure SD card lines have pull-up resistors in place.`

I will change back to SD library and wait a bit longer (much improvements can not be expected based on some threads found on the net).

schreibfaul1, 2021-02-12:
This is a bit problematic, PullUp on GPIO2 must only be active after booting.
Or alternatively burn the fuse: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#conflicts-between-bootstrap-and-sdio-on-dat2

biologist79, 2021-02-12:
@christi-ko 
The problem with GPIO2, that SD-boards available usually have a pullup-resistor soldered. You have to make sure to remove it. In parallel, you need to activate it by software. 
Like this one: https://github.com/biologist79/ESPuino/blob/master/pictures/Pullup-removal.jpg
It's this board: https://de.aliexpress.com/item/32867572635.html

If you've done so, you'll be able to flash without any problems and SD_MMC will work.

christi-ko, 2021-02-12:
Hi and thanks for the fast reply, OK is see the issue. I have a 2nd SD Card Reader an will try to do this later on. However i looked into SdFat a bit more and made a compare of time to list files and used Flash on my ESP32:

SD Library:
RAM:   [          ]   4.8% (used 15628 bytes from 327680 bytes)
Flash: [==        ]  19.6% (used 256691 bytes from 1310720 bytes)
Time to create a list (with new and delete functions) of 95 MP3 files which are distributed in 3 directories: ~12,2s

SdFat Library
RAM:   [          ]   5.0% (used 16364 bytes from 327680 bytes)
Flash: [==        ]  17.7% (used 232141 bytes from 1310720 bytes)
Time to create a list (with new and delete functions) of 95 MP3 files which are distributed in 3 directories: ~0,3s

If i will have success with the SD_MMC library i will update the analysis later on :-) ...

christi-ko, 2021-02-13:
Hi Wolle, 
this issue is not closed with the linked commit, i can't see any dependency maybe you closed the wrong one. This may be wontFix, but :-) ....
I would really like to see the SdFat Library working with audioIS2 (see results from pevious post showing the potential improvements). It's really worth i think.
regards
Christian

schreibfaul1, 2021-02-13:
Hi Christian,
Sorry, I erroneously linked to # 111, but # 112 is correct. Was a write error in the commit. Do you have success with SDFat?

christi-ko, 2021-02-13:
Hi Wolle,
Did you see the test results i achieved in listing files from my previous post:
https://github.com/schreibfaul1/ESP32-audioI2S/issues/111#issuecomment-778229807
I think this is a great improvement. I also tried to change your library to implement SdFat, but i am not good enough with c++ and don't understand all the things you did in your code ==> therefore i failed 👎 . 
Just worked on SD_MMC but was not able to get it stable running here. In my test system it works but not within my "radio" maybe a power supply issue. So i switched back to SD library for now, which is running very stable, but takes a long time.
regards
Christian

schreibfaul1, 2021-02-13:
I had never seen that before. SD_MMC is a little tricky. I have a board, GPIO2 is connected to ground via an LED, then no PullUp helps either, only burning the fuse helped. I did a test with 1MB of writing and reading:

SD_MMC ONE_WIRE
1048576 bytes read for 1027 ms
1048576 bytes written for 9934 ms

SD, SPI 26.6MHz
1048576 bytes read for 3220 ms
1048576 bytes written for 16823 ms

[SD_MMC.ino.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/5975830/SD_MMC.ino.txt)

clear-sky, 2021-02-13:
https://github.com/djuseeq/Ch376msc

christi-ko, 2021-02-15:
Hi, 
i just saw, that this there are many peoples having this issue with the compatability of the SdFat Library to the SD library. Looks like there is an wrapper which is made for the teensy, please the attached links: 

https://forum.pjrc.com/threads/64136-File-abstraction-and-SdFat-integration/page3?s=7c20005c17bc11e47e524ce3513e378f
https://github.com/PaulStoffregen/SD

Do you think this can be useful to solve the topic here?
regards
Christian

schreibfaul1, 2021-02-15:
Hi,
I looked at Paus Stoffregen's project. The global name 'FILE' is assigned again, which leads to ambiguities. I think this cannot be used without changes.

anp59, 2021-03-19:
Hi schreibfaul1, 
first of all, thank you for the very good ESP32-audioI2S library. 
Here an example of an SD player with the SdFat library (PlatformIO project). It works for SD-cards with FAT16/32 or exFat. 
[https://github.com/anp59/SD_test](url)
SD_libs serves as a minimal wrapper library. Note that the interface of the SdFat lib does not implement all methods of the SD lib/FS lib 1:1. 
In Audio::connecttoFS, few minor adjustments are necessary, since the SdFat uses the method File.getName().  
An additional change in the SdFat-Lib (and Audio.cpp) also allows the use of special characters in file names.
Perhaps the solution can serve as a basis for an integration of the SdFat-Lib.
regards
Andreas

christi-ko, 2021-03-19:
Hi Andreas,
can you check the link, for me it's not working.
regards
Christian

schreibfaul1, 2021-03-19:
Hi Andreas,
I had already tried to integrate SDFat, but unsuccessfully. But you have already done a lot of preparatory work, thank you very much for that. I will be testing in the next few days.
regards
Wolle

anp59, 2021-03-19:
Hi, 
sorry for the bad link. Here the correct one -> [https://github.com/anp59/SD_test.git](https://github.com/anp59/SD_test.git)
Andreas

schreibfaul1, 2021-03-21:
Hi Andreas,
I added the SD_Libs and made the changes you suggested in the source code. I think it works so far, SDFATFS_USED must be activated in Audio.h. The SdFat is not compatible with SD_MMC, SPIFFS .., there should be no reference to it. Please test the changes I have made, I would like to know if it can stay that way.
Wolle

anp59, 2021-03-21:
Hi Wolle,
the SD_Libs were intended as a test to be able to switch between the different FS implementations without code adaptation (at least in the Example). I'll have a look at the solution.  
Andreas

schreibfaul1, 2021-03-21:
I don't know how the SD_Libs differentiates between SD and SdFat. If both are active together I get a lot of error messages

anp59, 2021-03-21:
Hi,
for the use SdFat a global object SD_SDFAT is created. I had inserted a '#define SD SD_SDFAT' to continue using the name SD. This way, only one library can be included at a time.This is not absolutely necessary, but saves adjustments in Audio.h and Audio.cpp (e.g. ConnectToSD...). 
I have uploaded a version with an example (main.cpp) (branch dev):
[Example project for using SdFat](https://github.com/anp59/SD_test/tree/dev)
I renamed SD_libs to SD_SdFat and removed unnecessary code. In Audio.h, USE_SDFAT now controls whether SdFat is used. 

Andreas

christi-ko, 2021-04-07:
Can anyone tell me the status on this. Seems like there has been a version which included SDFat which has been removed afterwards. Do you think there will be a way to integrate or this seems to be too difficult?

Christian

en-ot, 2021-05-11:
Using SdFat solution with ESP32-audioI2S from anp59 works fine for me.
Also I recommend using SdFat beta, because it supports unicode.

schreibfaul1, 2021-05-12:
Hi en-ot,
yes, SdFat works fine in the example from anp59. Unfortunately, 'SdFat.h' cannot coexist with 'SD.h' in the same project. As a result, other libraries such as FTPServer, WebServer, TFT-Libs etc. would then no longer be able to work with the audioI2S Lib.

schreibfaul1, 2021-05-15:
There is an update for the implementation of SdFat. Please use the SdFat beta version, I added an example in the examples folder. Many thanks to anp59 for the preparatory work.

en-ot, 2021-05-17:
I've replaced the lib with new one in my project, It works better than before (from anp59).
But I can not use the lib as is, because of this lines in Audio.cpp:
#ifdef SDFATFS_USED
fs::SDFATFS SD_SDFAT;
#endif

I am using SdFat in other files, where I can not include Audio.h.
So now I use hybrid solution - SD_Libs from anp59 plus new lib without fs::SDFATFS SD_SDFAT, it works ok.
Maybe storing fs::SDFATFS SD_SDFAT in Audio.cpp is not so good idea? Or possibly I am doing something wrong :)

closed 2021-07-29 by stale[bot]

anp59, 2025-06-02:
Hi ShahzadaG,
yes, there is a new approach to use the SdFat library (without customizing the ESP32-audioI2S library):
https://github.com/anp59/SdFatFS

The SdFatFS library implements the functions of the Arduino FS interface based on the SdFat solution from Greimann (https://github.com/greiman/SdFat) and can therefore be used as an alternative to the SD or SD_MMC libraries in order to use additional SdFat functionalities in projects. 

For details, please take a look at the example PlayFilesOnSD-SdFatFS. It is suitable for all ESP32 or ESP32-S3 boards.
I have used the yellobyte board https://github.com/yellobyte/ESP32-DevBoards-Getting-Started/tree/main/boards/YB-ESP32-S3-AMP

szybkijanek, 2025-06-09:
Its my simple program, that integrate SD.h and SdFat.h in one sketch. Ofcourse works very fine on my ESP32 Wroover, need yet BT source choose.

[SD_SdFat_ESP32-audioI2S.txt](https://github.com/user-attachments/files/20653962/SD_SdFat_ESP32-audioI2S.txt)

commit cefe7c5c (2021-02-13): fix #111, start after audio_eof_stream event