schreibfaul1/ESP32-audioI2S #113: Cannot play MP3 any longer: ID3 version must be 3, skip all metadata

tueddy, 2021-02-13:
Hi,

first thanks for your great library!
The attached MP3 played fine before, now i get this messages and no playing:

```
info        : ID3 version=2.2
info        : ID3 version must be 3, skip all metadata
info        : ID3 framesSize=2440
info        : ID3 version=2.2
info        : ID3 version must be 3, skip all metadata
info        : ID3 framesSize=2440
info        : ID3 version=2.2
info        : ID3 version must be 3, skip all metadata
info        : ID3 framesSize=2440
info        : ID3 version=2.2
info        : ID3 version must be 3, skip all metadata
info        : ID3 framesSize=2440
info        : ID3 version=2.2
```
Here is a sample:

[Sample.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/5976189/Sample.zip)

What can i do do make it playing again?

schreibfaul1, 2021-02-13:
Hallo tueddy,
ich hatte die Prozedur für des Lesen der ID3 Metadaten erneuert, v2.4 hinzugefügt und v2.2 nicht weiter betrachtet weil die Version sehr selten ist. V2.2 ist wieder reaktiviert.

```
audio_info: stream ready
audio_info: ID3 framesSize=2440
audio_info: ID3 version=2.2
audio_id3data: Title/Songname/Content description: Alles gefunden! - und weitere Geschichten vom kleinen Raben Socke, Kapitel 1
audio_id3data: Lead artist(s)/Lead performer(s)/Soloist(s)/Performing group: Nele Moost & Annet Rudolph
audio_id3data: Track number/Position in set: 1/9
audio_id3data: Part of a set: 1/1
audio_id3data: Year: 2015
audio_id3data: Content type: Spoken & Audio
audio_id3data: Encoded by: iTunes 12.8.2.3
audio_info: syncword found at pos 0
audio_info: Channels=1
audio_info: SampleRate=44100
audio_info: BitsPerSample=16
audio_info: BitRate=80000
```
viele Grüße aus Berlin
Wolle

tueddy, 2021-02-13:
Hi Wolle,

Vielen Dank für den superduperschnellen Fix!
Ich habe neu ausgecheckt, clean & built durchgeführt, bekomme jetzt leider einige Compilerfehler, z.B

```
.pio\libdeps\lolin32\ESP32-audioI2S-master\src\Audio.cpp:461:19: error: 'CODEC_M4A' was not declared in this scope
         m_codec = CODEC_M4A;             
.pio\libdeps\lolin32\ESP32-audioI2S-master\src\Audio.cpp:740:9: error: 'm_ID3version' was not declared in this scope
         m_ID3version = *(data + 3);
..pio\libdeps\lolin32\ESP32-audioI2S-master\src\Audio.cpp:1014:43: error: 'bigEndian' was not declared in this scope
         size_t len = bigEndian(data + 3, 3);                                        ^
.pio\libdeps\lolin32\ESP32-audioI2S-master\src\Audio.cpp:1136:47: error: 'specialIndexOf' was not declared in this scope
             if(specialIndexOf(data, "ftyp", 10) != 4) {

```
Vermutlich fehlt hier noch etwas?
Schöne Grüße aus Bremen
Dirk

schreibfaul1, 2021-02-13:
da hab ich doch die Audio.h Datei vergessen, jetzt wird es funktionieren :-)

tueddy, 2021-02-13:
Perfekt, schnurrt wie 'ne Katze!

Dafür das Du "schreibfaul" bist ging der Fix ganz schön schnell!
Vielen Dank dafür & einen schönen Abend
Dirk

closed 2021-02-13 by tueddy