schreibfaul1/ESP32-audioI2S #115: Simple WiFI Radio Compile Error

rodo2020, 2021-02-16:
Hello,
I get this error when compiling:
'audio.Audio::connecttohost' does not have class type
Thanks for help.

kamil-m-m, 2021-03-07:
Hi,
I'v got another error:
Simple_WiFi_Radio:65:18: error: variable or field 'draw_button' declared void
 void draw_button(btns b)
  Thanks for help.

schreibfaul1, 2021-03-07:
Which IDE are you using? There are no problems with the Arduino IDE. Install the two libraries (audio and TFT) as a zip file and then create a new sketch. Then copy the content of 'Simple_WiFi_Radio.ino' into it, enter the WiFi SSID and PW, compile and upload it. And first set the partition scheme to 'Huge App' so that there is enough storage space.

kamil-m-m, 2021-03-07:
I am using Arduino 1.8.13 and arduino-esp32 from https://github.com/espressif/arduino-esp32 (fresh installation).  
Libraries ESP32-audioI2S-master.zip and ESP32-TFT-Library-ILI9486-master.zip added.  Partition scheme set to "Huge App", SSID and password set too.
Now I'v got different error:
`C:\Users\mkm\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp: In constructor 'Audio::Audio()':
C:\Users\mkm\Documents\Arduino\libraries\ESP32-audioI2S-master\src\Audio.cpp:142:17: error: 'class WiFiClientSecure' has no member named 'setInsecure'
`
My board is LILYGO TTGO T-Koala Wrover-B.

schreibfaul1, 2021-03-07:
It looks pretty good, now load the new ESP32 Arduino version into the boards manager and it will work for sure
[image]

closed 2021-06-09 by stale[bot]