schreibfaul1/ESP32-audioI2S #119: getAudioFileDuration sometimes returns length of previously playing file when I switch MP3 mid-playback

rfestag, 2021-02-24:
I have a use case where I play 3 files in succession:

File 1
Loop File 2 until some event
File 3.

In the case of File 1 and File 3, I am using the `getAudioFileDuration` and `getAudioCurrentTime` methods to display the progress of the file playback.

File 1 always works as expected, and is always preceded by silence (no file playing back).
File 3 is always played by interrupting File 2 playback.
I wait until I get a non-zero value for the duration and current time (i.e., presumably they are both already reset). I consistently see a time where those values are 0 before the current time is non-zero.

Unfortunately, sometimes when I get the File 3 duration, I get the duration from File 2. It may work fine the first time, but the second time and so on after seem to always return the duration of File 2. Any ideas why this might be the case?

rfestag, 2021-02-24:
Update: I have noticed that, in the cases where the wrong duration is calculated, I'm not seeing the ID3 headers being printed. I currently have this:

```c
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
```
The intersting part is it does seem to know the right file, and stops playing the file at the expected time (about 1.2s). But Something wonky may be happening when changing from a looped file.

Here is what I do to change my file: 
```c
void play(const char *file, bool replay) {
  audio.setFileLoop(false);
  audio.connecttoFS(SPIFFS, file);
  audio.setFileLoop(replay);
  Serial.print("Playing ");
  Serial.println(file);
}
```

schreibfaul1, 2021-02-24:
Presumably an (mp3) file has no or a damaged ID3 header. So far I have calculated the audioDataSize in the header (m_audioDataSize = m_contentlength - id3Size). If no header is found, I now assume that m_audioDataSize = m_contentlength. Hope it works like that.

rfestag, 2021-02-24:
Awesome, looks like this is working more reliably now. Thanks for the quick reply!

closed 2021-02-24 by rfestag