schreibfaul1/ESP32-audioI2S #121: How to smoothly change files

rfestag, 2021-02-25:
I am using an Adafruit Propmaker Featherwing (https://www.adafruit.com/product/3988) and an ESP32 feather wing. In general, things are working great. However, whenever I start a new file for playback, I get some clicking and a little dead space.

I am trying to use this to build a prop that reacts to button presses or the accelerometer, and the clicking is kind of annoying. I'm not entirely sure how to identify what might be the issue. I have tried a couple of things on my end with no luck:

1. Explicitly stopping the playback (`stopSong()`)
2. Updating the `connecttoFS` method to check to see if a file is running, and if so I only do a partial reset (I skip clearing/freeing buffers, but I reset all variables). 

Neither seems to have changed the behavior.  Any idea what I can/should do to identify the issue?

schreibfaul1, 2021-02-26:
This is a problem when the I2S driver is started. I guess you mean a short dropout at the beginning of the file. This could now work better with the update.

rfestag, 2021-02-26:
So, to start off this new version does seem to behave slightly better (I used to here more clicks). That said, it is still there. It's not so much a dropout and more of a clicking sounds. I have a sample of the behavior:
https://drive.google.com/file/d/1dJ_KBAbwwR_rgLfX3ID4q2XOP2IfO0vy/view?usp=sharing

You can hear a click around 3s, 5s, 8s, 9s, 11s, and 14s as different files are iterated through.

There is a very slight silence when the file loops, but no click.

closed 2021-06-09 by stale[bot]