schreibfaul1/ESP32-audioI2S #123: audioFileSeek() not working?

biologist79, 2021-02-26:
Hi,

just tried around a bit with audioFileSeek().
audioFileSeek(4) leads to:

nfo        : framesize is 0, start decoding again
info        : syncword found at pos 86
info        : syncword found at pos 0
info        : framesize is 0, start decoding again
info        : syncword found at pos 73
info        : syncword found at pos 0
info        : framesize is 0, start decoding again
info        : syncword found at pos 180
info        : syncword found at pos 0
info        : MP3 decode error -9 : INVALID_HUFFCODES
info        : syncword found at pos 226
info        : syncword found at pos 0
info        : framesize is 0, start decoding again
info        : syncword found at pos 6
info        : syncword found at pos 0
info        : MP3 decode error -9 : INVALID_HUFFCODES
info        : syncword found at pos 51
info        : syncword found at pos 0
info        : framesize is 0, start decoding again
info        : syncword found at pos 0
info        : MP3 decode error -8 : INVALID_SCALEFACT
info        : syncword found at pos 409
info        : syncword found at pos 0
info        : MP3 decode error -2 : MAINDATA_UNDERFLOW
info        : syncword found at pos 0
info        : Channels: 2
info        : SampleRate: 44100
info        : BitsPerSample: 16
info        : BitRate: 128000

audioFileSeek(-4) leads to:
info        : framesize is 0, start decoding again
info        : syncword found at pos 51
info        : syncword found at pos 0
info        : MP3 decode error -9 : INVALID_HUFFCODES
info        : syncword found at pos 0
info        : Channels: 2
info        : SampleRate: 44100
info        : BitsPerSample: 16
info        : BitRate: 128000

However, in both cases the music is still playing. 
Hint: I applied both commands when the music was already playing.

At last a question: What has to be set to go back to "normal" playmode? 0?

Many thanks in advamce.

biologist79, 2021-02-26:
Also tried it with +/- 1 and +/- 2. But the result is the same.

schreibfaul1, 2021-02-26:
Hi biologist79,
audioFileSeek() is not perfect, mp3 and aac files are divided into frames (often between 600 ... 1400 bytes long). At the beginning the frames have a synconword and mostly afterwards information about the length of the frame, sample rate etc. With audioFileSeek() the pointer is set to a position that rarely points to a syncword. Then the decoder looks for the beginning of the next usable frame. I certainly have something to do here.

regards
Wolle

schreibfaul1, 2021-02-28:
I have created a new function, setTimeOffset (int sec), with which you can rewind and fast forward in a local audio file, setTimeOffset(-30) means 30 seconds back. This works with mp3 and wav files. And the mp3 decoder is now looking for the sync word first.

biologist79, 2021-03-01:
Hi Wolle,

will test this. Thank you!

closed 2021-06-09 by stale[bot]

0x0fe, 2023-03-19:
@biologist79  did you get this function working during playback? basically to do playback forward and playback reverse. I tried both setTimeOffset and audioFileSeek() but none worked during playback. audioFileSeek() will cause the currentbuffer to be played in loop 3 or 4 times and then stop playback all together.

schreibfaul1, 2023-03-19:
With the code you can jump forwards or backwards via a terminal:
````c++
void loop(){
    audio.loop();
	static uint32_t t = millis() + 1000;
	if (t < millis()) {
		t = millis() + 1000;
		log_i("currentTime %is, fileDuration %is", audio.getAudioCurrentTime(),
				audio.getAudioFileDuration());
	}
        if(Serial.available() > 0){
            String st = Serial.readString();
            log_i("%s", st);
           audio.setTimeOffset(st.toInt());
    }
}
````

0x0fe, 2023-03-19:
well, that is what i do in my test code, and unfortunately it just repeats the current buffer indefinitely as soon as setTimeOffset is called. Maybe this is because i used SD_MMC as source?

fizicko, 2023-10-09:
hi, schreibfaul1.
Please tell me how can I change the playback speed of a mp3 file? Or maybe this is possible with other file types? The audioFileSeek() function does not work.

pitlochry, 2023-12-09:
Hi @fizicko, audioFileSeek seems not to work, which is a very huge pity. 

The last time I tried it in August 23 on esp32 (espuino), where setting e.g. audio.audioFileSeek(1.4) the speed was faster than normal, but extremely choppy. Do you observe the same behaviour?

See https://github.com/schreibfaul1/ESP32-audioI2S/issues/29

Perhaps this will eventually be fixed, it would be a great avail to this library.