schreibfaul1/ESP32-audioI2S #126: Guru Meditation Error -> not enough memory to allocate mp3decoder buffers

christi-ko, 2021-03-02:
Some Streams seem to work, some others i am getting this problem. I am already working on my code to save more memory, and i think i will get some more 10.000bytes back by improving it. 2 Questions  about this behavour

I was not able to identfy why some streams seems to need more buffer. Is this behaviour according to your expectations

1.) Is this according to your expectations?
2.) What is your advice to have as a min. free Heap available for  the library to play webstreams

Getting a core panic due to low buffer available:
**1st Example with a bit higher RAM usage**
`DEBUG:        buffers freed, free Heap: 162896 bytes
DEBUG:        Connect to new host: "streambbr.ir-media-tec.com/berlin/mp3-128/vtuner_web_mp3/"
DEBUG:        Connect to "streambbr.ir-media-tec.com" on port 80, extension "/berlin/mp3-128/vtuner_web_mp3/"
DEBUG:        Connected to server
DEBUG:        HTTP/1.1 302 Found
DEBUG:        access-control-allow-origin: *
DEBUG:        content-type: audio/mpeg, format is mp3
DEBUG:        MP3Decoder has been initialized, free Heap: 136924 bytes
DEBUG:        redirect to new host "http://irmedia.streamabc.net/irm-bbrberlinclub-mp3-128-4574783?sABC=603r4r52%230%2380s2q3r7spop7qr7op166460pp8297q9%23ighare_jro_zc3&=&amsparams=playerid:vtuner_web_mp3;skey:1614696018"
DEBUG:        buffers freed, free Heap: 161920 bytes
DEBUG:        Connect to new host: "http://irmedia.streamabc.net/irm-bbrberlinclub-mp3-128-4574783?sABC=603r4r52%230%2380s2q3r7spop7qr7op166460pp8297q9%23ighare_jro_zc3&=&amsparams=playerid:vtuner_web_mp3;skey:1614696018"
DEBUG:        Connect to "irmedia.streamabc.net" on port 80, extension "/irm-bbrberlinclub-mp3-128-4574783?sABC=603r4r52%230%2380s2q3r7spop7qr7op166460pp8297q9%23ighare_jro_zc3&=&amsparams=playerid:vtuner_web_mp3;skey:1614696018"
DEBUG:        Connected to server
DEBUG:        HTTP/1.1 200 OK
DEBUG:        content-type: audio/mpeg, format is mp3
DEBUG:        MP3Decoder has been initialized, free Heap: 135136 bytes
DEBUG:        server: QuantumCast Streamer (Version v1.8.7)
DEBUG:        vary: Origin
DEBUG:        x-quantumcast-channelkey: bbradio-berlinclub
DEBUG:        ice-audio-info: samplerate=44100;bitrate=128;channels=2
DEBUG:        icy-description: BB RADIO - Berlin Club
DEBUG:        icy-name: BB RADIO - Berlin Club
DEBUG:        Switch to DATA, metaint is 8192
DEBUG:        inputbuffer is being filled
DEBUG:        stream ready
DEBUG:        buffer filled in 46 ms
DEBUG:        syncword found at pos 0
DEBUG:        Channels: 2
DEBUG:        SampleRate: 44100
DEBUG:        BitsPerSample: 16
DEBUG:        BitRate: 128000
DEBUG:        StreamTitle="BB RADIO - Berlin Club"
DEBUG:        StreamTitle=""
DEBUG:        buffers freed, free Heap: 161008 bytes
DEBUG:        Connect to new host: "https://stream.rockantenne.de/rockantenne/stream/mp3"
DEBUG:        Connect to "stream.rockantenne.de" on port 443, extension "/rockantenne/stream/mp3"
DEBUG:        SSL has been established, free Heap: 117888 bytes
DEBUG:        HTTP/1.0 302 Found
DEBUG:        access-control-allow-origin: *
DEBUG:        redirect to new host "https://s5-webradio.rockantenne.de/rockantenne/stream/mp3"
DEBUG:        buffers freed, free Heap: 161524 bytes
DEBUG:        Connect to new host: "https://s5-webradio.rockantenne.de/rockantenne/stream/mp3"
DEBUG:        Connect to "s5-webradio.rockantenne.de" on port 443, extension "/rockantenne/stream/mp3"
DEBUG:        SSL has been established, free Heap: 117764 bytes
DEBUG:        HTTP/1.0 200 OK
DEBUG:        access-control-allow-origin: *
DEBUG:        content-type: audio/mpeg, format is mp3
[E][mp3_decoder.cpp:1527] MP3Decoder_AllocateBuffers(): not enough memory to allocate mp3decoder buffers
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c46c  PS      : 0x00060430  A0      : 0x8015fcbd  A1      : 0x3ffb1a70  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001b20  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x000001b2  A8      : 0x8000beca  A9      : 0x3ffb1a00  
A10     : 0x3ffb80c0  A11     : 0x3ffdb49c  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060620  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x000001b1  
ELF file SHA256: 0000000000000000
Backtrace: 0x4000c46c:0x3ffb1a70 0x4015fcba:0x3ffb1a80 0x4015fe27:0x3ffb1aa0 0x400d69ec:0x3ffb1ad0 0x400dd189:0x3ffb1af0 0x400dd57f:0x3ffb1d20 0x400de0ca:0x3ffb1f60 0x400d28ef:0x3ffb1f90 0x400e3add:0x3ffb1fb0 0x40089c0e:0x3ffb1fd0
Rebooting...
`

**2nd Example with less full RAM opening ROCKANTENNE (one of my streams that requires more RAM)**
`DEBUG:        buffers freed, free Heap: 216148 bytes
DEBUG:        Connect to new host: "https://stream.rockantenne.de/rockantenne/stream/mp3"
DEBUG:        Connect to "stream.rockantenne.de" on port 443, extension "/rockantenne/stream/mp3"
DEBUG:        SSL has been established, free Heap: 172848 bytes
DEBUG:        HTTP/1.0 302 Found
DEBUG:        access-control-allow-origin: *
DEBUG:        redirect to new host "https://s2-webradio.rockantenne.de/rockantenne/stream/mp3"
DEBUG:        buffers freed, free Heap: 216288 bytes
DEBUG:        Connect to new host: "https://s2-webradio.rockantenne.de/rockantenne/stream/mp3"
DEBUG:        Connect to "s2-webradio.rockantenne.de" on port 443, extension "/rockantenne/stream/mp3"
DEBUG:        SSL has been established, free Heap: 172712 bytes
DEBUG:        HTTP/1.0 200 OK
DEBUG:        access-control-allow-origin: *
DEBUG:        content-type: audio/mpeg, format is mp3
DEBUG:        MP3Decoder has been initialized, free Heap: 151440 bytes
DEBUG:        server: streaMonkey streaming Server Native
DEBUG:        sessionid: 603e4cbea77a6c4556836177
DEBUG:        icy-audio-info: ice-channels=2;ice-samplerate=44100;ice-bitrate=128
DEBUG:        icy-description: ROCK ANTENNE
DEBUG:        icy-name: ROCK ANTENNE
DEBUG:        icy-url: http://www.rockantenne.de
DEBUG:        date: Tue, 02 Mar 2021 14:33:35 GMT
DEBUG:        Switch to DATA, metaint is 16000
DEBUG:        inputbuffer is being filled
DEBUG:        stream ready
DEBUG:        buffer filled in 57 ms
DEBUG:        syncword found at pos 0
DEBUG:        Channels: 2
DEBUG:        SampleRate: 44100
DEBUG:        BitsPerSample: 16
DEBUG:        BitRate: 128000
DEBUG:        StreamTitle="ROCK ANTENNE - Rock Nonstop!"
DEBUG:        StreamTitle="ROCK ANTENNE - Rock Nonstop"
`

schreibfaul1, 2021-03-03:
Hello christi-ko,
It is so that the decoder releases the memory again after use. Free Heap shows the available RAM; this is not the maximum contiguous memory area, especially when the heap is fragmented. With SSL connections it gets tight. However, the ESP32 must not crash in this case. I'll fix that with the next update. If possible, you should always give preference to unencrypted connections with regard to the memory used, they will also load faster. I'm trying to put the decoder's buffers in the PSRAM. Then the MP3 decoder only uses 300 bytes in the heap and there is plenty of space for your own extensions. Initial attempts at this are promising.

christi-ko, 2021-03-03:
Hi Wolle,

thank you again for your great support. Now i understand it a bit better and i try to change all my stream to a http server not to use https:. Also i changed my code to use less memory, which is a general worthy target ...
My ESP32-S which i use has no PSRAM i guess. But it should work even without with this changes. So i will close this issue as you anyhow work on the mentioned improvements.

regards
Christian

closed 2021-03-03 by christi-ko

ardyesp, 2025-12-27:
I tried using V3.2.1 with an ESP32 board without PSRAM. While MP3 streams work fine, the AAC streams throw an exception. The AAC decoder in ```libfaad/neaacdec.cpp``` uses ```ps_malloc``` to allocate memory, which will fail.
The highest version with working AAC stream is version 3.0.12.