schreibfaul1/ESP32-audioI2S #131: Some streams do not work anymore..

CelliesProjects, 2021-03-10:
Hi Wolle.

I just got even with master coming from 06821fd.

After this update some stations stopped playing and these always error out on the `connecttohost` call.

Example:
```c++
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

#define I2S_DOUT      22
#define I2S_BCLK      21
#define I2S_LRC       26

Audio audio;

String ssid =     "xxx";
String password = "xxx";

void setup() {
  // put your setup code here, to run once:
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED) delay(1500);
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(21); // 0...21

  //audio.connecttohost("http://icecast.omroep.nl/radio1-bb-mp3");
  audio.connecttohost("http://stream.afc.fm:8504/listen.pls");
}

void loop() {
  audio.loop();
}
```
After a while the call returns with `[E][WiFiClient.cpp:258] connect(): socket error on fd 54, errno: 113, "Software caused connection abort"`

CelliesProjects, 2021-03-10:
Nvm. I see the stream has problems in every player.

schreibfaul1, 2021-03-10:
Hi Cellie,
the icecast stream works, afc (Amsterdan Funk Channel?) seems to be offline.
Many greetings from Berlin
Wolle

CelliesProjects, 2021-03-12:
Everything back to normal.

I guess it was a DNS problem at my ISP.

closed 2021-03-12 by CelliesProjects