schreibfaul1/ESP32-audioI2S #133: TTGO TM Music - playing mp3 from SD results in mono audio

gabry1609, 2021-03-15:
Hello guys 👍🏻 

first thanks for your great work. Actually I'm experincing a problem with my **TTGO TM Music** board. By using your sketch, I can hear the song correctly, but unfortunately only from the left channel. I'm using a pair of Samsung hearphones and I can start hearing something as soon as I hit any of the buttons on the earphone's cord. I've tried also with other earphones without buttons but I'm having the same issue. **Audio plays only on the left channel**, even if I specified forceMono to false.

Can you please help? Code attached. Thanks in advance :)

```
//#include "Arduino.h"
//#include "WiFi.h"
#include "Audio.h"
//#include "SD.h"
//#include "FS.h"

// Digital I/O used
#define SD_CS 13
#define SPI_MOSI 15
#define SPI_MISO 2
#define SPI_SCK 14
#define I2S_DOUT 19 //DIN
#define I2S_BCLK 26 //BCK
#define I2S_LRC 25 //LCK

Audio audio;

void setup() {
  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);

  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  Serial.begin(115200);
  Serial.println("Start SD Test");
  if (!SD.begin(SD_CS)) {
    Serial.println("Card Mount Failed");
  }
  Serial.println("End SD test");

  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.forceMono(false);
  audio.setVolume(20); // 0...21
  Serial.println("Start song test");
  audio.connecttoFS(SD, "/320k_test.mp3");
  Serial.println("End song test");
}

void loop()
{
  audio.loop();
}
```

schreibfaul1, 2021-03-15:
Basically, in the I2S protocol, the right and left channels are transmitted alternately, even with mono. I suspect that the line for LCR (or sometimes called WS) is on a different pin on your board.

gabry1609, 2021-03-15:
Thank you for your suggestion. I'll check the pinout, taught it was right because I was looking at the one from [this old post](https://github.com/schreibfaul1/ESP32-audioI2S/issues/87), that gave me the sound on the Left channel. Unfortunately there isn't so much documentation about this board, will try to do my best and let you know tomorrow.

Also I've found [this official example](https://github.com/karawin/Ka-Radio32#gpio-definition) from the board manufacturer, and as you can see (at the 4th comment from the end) the pinout is a bit different in this one, but only for the data part. Could this be the problem? I'm not quite sure...

gabry1609, 2021-03-15:
My bad, friend. After closely looking at the board pcb, I found out that the connection with the right channel pin was a little bit loose. After soldering, the problem stopped :) Thanks again for your great work.
[image]

closed 2021-03-15 by gabry1609