schreibfaul1/ESP32-audioI2S #137: slow stream, dropouts are possible

yatatyork, 2021-03-25:
Hello Mr.
I can connecthost https://mp3.ffh.de/ffhplus/hq80er.mp3 
But when it print title and url about songs once 
then always show  "slow stream, dropouts are possible"  about 1minute then lost 
(WiFi connect Personal Hotspot)
Is it just poor Internet problem?
I will go to school and test school wifi
[image]
[image]

schreibfaul1, 2021-03-25:
It's the same with me too. Maybe a temporary problem or a bad connection. Please try http://mp3.ffh.de/ffhchannels/80er.aac
Radio Browser https://www.radio-browser.info/#!/ is a good address to find alternative URLs

yatatyork, 2021-03-26:
Thank for your reply!
I will try it and report my result

closed 2021-06-09 by stale[bot]