schreibfaul1/ESP32-audioI2S #141: Radio is streaming but not getting audio

playground, 2021-04-03:
I finally got the components in the mail.  Hooked the up according to instructions but not getting any sound.
What am I missing?

info        buffers freed, free Heap: 226048 bytes
info        Connect to new host: "http://mp3.ffh.de/radioffh/hqlivestream.mp3"
info        Connect to "mp3.ffh.de" on port 80, extension "/radioffh/hqlivestream.mp3"
info        Connected to server in 464 ms
info        HTTP/1.0 200 OK
bitrate     128000
info        icy-description: Unspecified description
info        icy-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
info        icy-url: https://www.ffh.de
icyurl      https://www.ffh.de
info        instance-id: 247d319d0e6cd2e25d9d21715292c8af
info        server: AIS Streaming Server 8.4.3
info        icy-name: HIT RADIO FFH
station     HIT RADIO FFH
info        content-type: audio/mpeg, format is mp3
info        MP3Decoder has been initialized, free Heap: 200136 bytes
info        Switch to DATA, metaint is 16000
lasthost    mp3.ffh.de/radioffh/hqlivestream.mp3
info        inputbuffer is being filled
info        stream ready
info        buffer filled in 467 ms
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 128000

[image]

schreibfaul1, 2021-04-03:
The log looks good. I use the same DAC. Is SCK connected to ground?

playground, 2021-04-04:
Hi, yes SCK is connected to ground.

playground, 2021-04-10:
@schreibfaul1 any idea why I'm not getting audio?  

I tried with Xtronical, I'm able to get audio with this setup
#include <Arduino.h>
#include "SoundData.h"
#include "XT_DAC_Audio.h"
XT_Wav_Class Sound(sample);                                          
XT_DAC_Audio_Class DacAudio(25,0);    
uint32_t DemoCounter=0;               
void setup() {
  Serial.begin(115200);               
}
void loop() {
  DacAudio.FillBuffer();                
  if(Sound.Playing==false)       
    DacAudio.Play(&Sound);       
  Serial.println(DemoCounter++);        
}

What I'm trying to accomplish is to be able to store the audio in SPIFF that I got from https://texttospeech.googleapis.com/v1/text:synthesize.  And play the audio by reading from SPIFF.  Any suggestion?

[image]

schreibfaul1, 2021-04-10:
Hi playground,
I assume the Xtronical SW uses the internal DAC. This is also possible, add the line in Setup():
audio.setInternalDAC(true);
Maybe there is a problem with the I2S bus?

playground, 2021-04-11:
Thanks @schreibfaul1, did something change recently.  I'm just now getting these build errors 

PACKAGES: 
 - framework-arduinoespressif32 3.10005.210308 (1.0.5) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32-audioI2S-master> 2.0.0+sha.783f67c
|   |-- <FFat> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SD(esp32)> 1.0.5
|   |   |-- <FS> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SD_MMC> 1.0
|   |   |-- <FS> 1.0
|   |-- <SPI> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pio/build/audio/src/main.cpp.o
src/main.cpp:14:1: error: 'Audio' does not name a type
 Audio audio;
 ^
src/main.cpp: In function 'void setup()':
src/main.cpp:22:5: error: 'SPI' was not declared in this scope
     SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
     ^
src/main.cpp:24:5: error: 'SD' was not declared in this scope
     SD.begin(SD_CS);
     ^
src/main.cpp:29:5: error: 'audio' was not declared in this scope
     audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
     ^
src/main.cpp: In function 'void loop()':
src/main.cpp:44:5: error: 'audio' was not declared in this scope
     audio.loop();
     ^
Compiling .pio/build/audio/lib889/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/audio/lib889/WiFi/WiFiScan.cpp.o
Compiling .pio/build/audio/lib889/WiFi/WiFiServer.cpp.o
Compiling .pio/build/audio/lib889/WiFi/WiFiUdp.cpp.o
Compiling .pio/build/audio/lib2dc/WiFiClientSecure/WiFiClientSecure.cpp.o
Compiling .pio/build/audio/lib2dc/WiFiClientSecure/ssl_client.cpp.o
Compiling .pio/build/audio/lib401/ESP32-audioI2S-master/Audio.cpp.o
Compiling .pio/build/audio/lib401/ESP32-audioI2S-master/aac_decoder/aac_decoder.cpp.o
Compiling .pio/build/audio/lib401/ESP32-audioI2S-master/flac_decoder/flac_decoder.cpp.o
Compiling .pio/build/audio/lib401/ESP32-audioI2S-master/mp3_decoder/mp3_decoder.cpp.o
*** [.pio/build/audio/src/main.cpp.o] Error 1
Archiving .pio/build/audio/libFrameworkArduinoVariant.a
Indexing .pio/build/audio/libFrameworkArduinoVariant.a
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp: In member function 'int Audio::read_M4A_Header(uint8_t*, size_t)':
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1439:18: warning: variable 'atomName' set but not used [-Wunused-but-set-variable]
             char atomName[5];
                  ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1522:80: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t {aka unsigned int}' [-Wformat=]
             sprintf(chbuf, "Sampling Frequency: %lu",samplingFrequencies[sRate]);
                                                                                ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1522:80: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t {aka unsigned int}' [-Wformat=]
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1485:21: warning: unused variable 'len_of_OD' [-Wunused-variable]
             uint8_t len_of_OD  = *(pos + 12); // length of this OD (which includes the next 2 tags)
                     ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1486:21: warning: unused variable 'len_of_ESD' [-Wunused-variable]
             uint8_t len_of_ESD = *(pos + 20); // length of this Elementary Stream Descriptor
                     ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1532:21: warning: unused variable 'dependsOnCoreCoder' [-Wunused-variable]
             uint8_t dependsOnCoreCoder  = (ASC & 0x02);
                     ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1533:21: warning: unused variable 'extensionFlag' [-Wunused-variable]
             uint8_t extensionFlag       = (ASC & 0x01);
                     ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp: In member function 'int Audio::read_OGG_Header(uint8_t*, size_t)':
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1622:19: warning: variable 'headerSize' set but not used [-Wunused-but-set-variable]
     static size_t headerSize;
                   ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:1624:19: warning: variable 'f_lastMetaBlock' set but not used [-Wunused-but-set-variable]
     static bool   f_lastMetaBlock;
                   ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp: In member function 'bool Audio::playSample(int16_t*)':
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:3301:13: warning: unused variable 'sample1' [-Wunused-variable]
     int16_t sample1[2]; int16_t* s1;
             ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:3301:34: warning: unused variable 's1' [-Wunused-variable]
     int16_t sample1[2]; int16_t* s1;
                                  ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:3302:34: warning: unused variable 's2' [-Wunused-variable]
     int16_t sample2[2]; int16_t* s2 = sample2;
                                  ^
.pio/libdeps/audio/ESP32-audioI2S-master/src/Audio.cpp:3303:34: warning: unused variable 's3' [-Wunused-variable]
     int16_t sample3[2]; int16_t* s3 = sample3;
                                  ^

closed 2021-06-18 by stale[bot]